mavlink/MAVSDK-Swift #15: Carthage setup is not working

byuarus, 2018-03-30:
[image]

Carthage fails to build DroneCore-Swift. It won't create the `Carthage/Build/iOS/Dronecode_SDK_Swift.framework`

I'm using:
- Sierra MacOS 10.12.6
- Xcode 9.2 Build version 9C40b
- Carthage version 0.29.0

Step to reproduce:

- create a new iOS single view app project.
- create the file Cartfile with code:
`# Require the iOS framework of DroneCore SDK`
`github "dronecore/DroneCore-Swift" `
- run Carthage in your app's repository
`carthage update`

byuarus, 2018-04-03:
Thank you, it fixed the `carthage update`! I got the `Carthage/Build/iOS/Dronecode_SDK_Swift.framework`. I have access to the Action class from my code.

Then I did [these steps](https://github.com/dronecore/DroneCore-Swift#add-the-framework-into-your-project):

> ## Add the framework into your project
> - Open Project Settings -> General
> - Find Embedded Binaries and press +
> - Click Other, go one folder up, and select Carthage/Build/iOS/Dronecode_SDK_Swift.framework.
> - Do "Product Clean" and "Product Build"

Now I have different issues:
- When I select simulator as my target device I'm getting this error when I try to build and run the app:
`dyld: Library not loaded: @rpath/RxSwift.framework/RxSwift
  Referenced from: /Users/dima/Library/Developer/CoreSimulator/Devices/5F589720-4CA4-4B85-9279-5E20E5E7A13D/data/Containers/Bundle/Application/601057E6-60DA-4AB7-ADDF-B90F1AC662ED/testCarthageDronecode.app/Frameworks/Dronecode_SDK_Swift.framework/Dronecode_SDK_Swift
  Reason: image not found`

- When I select an iPad I'm getting this error on the build stage:
`ld: '/Users/dima/Work/DroneCode&Yuneec/testCarthageDronecode/testCarthageDronecode/Carthage/Build/iOS/Dronecode_SDK_Swift.framework/Dronecode_SDK_Swift' does not contain bitcode. You must rebuild it with bitcode enabled (Xcode setting ENABLE_BITCODE), obtain an updated library from the vendor, or disable bitcode for this target. file '/Users/dima/Work/DroneCode&Yuneec/testCarthageDronecode/testCarthageDronecode/Carthage/Build/iOS/Dronecode_SDK_Swift.framework/Dronecode_SDK_Swift' for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

----------

Also, it is unclear what should I do in [the next step](https://github.com/dronecore/DroneCore-Swift#clean-up-the-download-automatically-done-by-the-build)

> ## Clean up the download automatically done by the build
> Dependencies are needed to build Dronecode_SDK_Swift.framework. The script download-dependencies.sh is downloading a zip containing them. To remove this download:
> 
> Open the project Dronecode-SDK-Swift in XCode and do the following:
> 
> - Open Project Settings -> Build Phases
> - Open Run Script tab
> - Put a # before ./download-dependencies.sh or just delete the runscript

- Do I need to manually run `download-dependencies.sh` or it will run automatically? 
- I didn't find `Run Script` tab in `Project Settings -> Build Phases`. Do I need manually add `Run Script` tab and paste the script or the Carthage setup isn't working properly?

Thank you for your help. I hope my questions and comments will help to improve the documentation and the setup process.

JonasVautherin, 2018-04-03:
The first error (`dyld: Library not loaded: @rpath/RxSwift.framework/RxSwift`) seems like the library has not been copied during the installation (`@rpath` is where it looks to load the library at runtime). @ayameMBS: shouldn't he add all the dependencies in his framework as well (i.e. in the "Embedded Binaries")? That's what I do, at least...

For the second error, I think you should disable the bitcode in the "Build Settings" (search for bitcode there). We could try to enable bitcode (all the deps have to be built with the bitcode, not sure what it implies), but I don't think it will make a difference if you disable it for your app.

Thanks for the comments! We will definitely take them into account for the documentation :-).

ayameMBS, 2018-04-03:
@JonasVautherin yes exactly. @byuarus you need also to add all the frameworks in the folder /Users/marjo/testCarthage/Carthage/Checkouts/DroneCore-Swift/Source/Dronecode-SDK-Swift/Dronecode-SDK-Swift/libs/dronecode-sdk-swift-deps-latest

And sure thanks for the comments ^^

JonasVautherin, 2018-05-29:
Closing in favor of #47.

closed 2018-05-29 by JonasVautherin