mavlink/MAVSDK-Swift #21: [Cocoa pod] Install error

nguyenvietvu, 2018-04-09:
I run "pod install" at
**/DroneCore-Swift-master/Sample Code/Swift Sample Code**
 but error occured
Please help me to solve it, pod's log below:

Analyzing dependencies
Downloading dependencies
Installing DroneCore-Swift (0.0.1)

[!] Error installing DroneCore-Swift
[!] /usr/bin/curl -f -L -o /var/folders/k4/5cpcyfbx6nd7y0fdvb6rqhz80000gn/T/d20180409-28514-1ibv0sn/file.zip https://s3.eu-central-1.amazonaws.com/dronecode-sdk/dronecore-swift-prototype.zip --create-dirs --netrc-optional

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 403 Forbidden

JonasVautherin, 2018-04-09:
Hello @nguyenvietvu!

The link is outdated, indeed. @ayameMBS or I will fix this soon. Please try again by the end of the week and let us know how it goes! :-)

JonasVautherin, 2018-04-13:
Hi @nguyenvietvu! So the cocoapod config is still not fixed, sorry about that.

However, if you want to give it a shot, you can try the manual install :-). Note that the Swift SDK is not yet feature-complete.

JonasVautherin, 2018-04-19:
@nguyenvietvu, I believe @ayameMBS fixed the bug [here](https://github.com/dronecore/DroneCore-Swift/pull/26).

Can you update, try again, and let us know if it works for you?

JonasVautherin, 2018-05-29:
Closing due to inactivity.

closed 2018-05-29 by JonasVautherin