mavlink/MAVSDK-Swift #50: Swift Version conflict in XCode 9.3

julianoes, 2018-05-30:
I tried to compile and run the example in order to test the live stream with the E90.

Unfortunately, I get the Swift version error:
```
/Users/julianoes/src/DroneCore-Swift/SampleCode/SwiftSampleCode/DronecodeSDKSwiftDemo/Controllers/TelemetryViewController.swift:10:8: Module compiled with Swift 4.0.3 cannot be imported in Swift 4.1: /Users/julianoes/src/DroneCore-Swift/SampleCode/SwiftSampleCode/Pods/DroneCore-Swift/Dronecode_SDK_Swift.framework/Modules/Dronecode_SDK_Swift.swiftmodule/arm64.swiftmodule
```

Could it be that `Dronecode_SDK_Swift.framework` is still built using XCode 9.2 or earlier? Or is something wrong with my setup/config?

JonasVautherin, 2018-05-30:
Are you using cocoapods? Did you run `pod update` and/or `pod install`?

I had the same issue because I was on 0.0.2, and the latest version is 0.0.3.

julianoes, 2018-05-31:
Aha yes I had only done `pod install`, `pod update` did the trick.

```
Installing DroneCore-Swift 0.0.3 (was 0.0.2)
Installing MobileVLCKit 3.1.0 (was 3.0.2)
```

Now I get:
```
ld: framework not found Pods_DronecodeSDKSwiftDemo
```

Thanks!

JonasVautherin, 2018-05-31:
Did you open the workspace then? I always make the mistake to open the xcodeproj, and then I miss the dependencies :sweat_smile:

julianoes, 2018-06-01:
Ah, thanks. I'll try that next.

LorenzMeier, 2018-06-01:
Do we have good documentation with Screenshots on how to do that?

JonasVautherin, 2018-06-04:
> Do we have good documentation with Screenshots on how to do that?

[Here](https://github.com/dronecore/dronecore-swift#cocoapods) are the installation instructions, and [here](https://github.com/dronecore/dronecore-swift#cocoapods-1) the update instructions. And that's all standard CocoaPods, I believe. 

The thing is that we are not so used to CocoaPods, so we have to get used to it ;-).

julianoes, 2018-06-07:
Ok once I opened the workspace it was all fine, thx.

closed 2018-06-07 by julianoes