mavlink/MAVSDK-Swift #56: Start/stop video stream are not working

JonasVautherin, 2018-06-11:


JonasVautherin, 2018-06-13:
There is a startVideo() and stopVideo() in the Camera.swift file.

Dmitry was not sure what it was supposed to do, but it seemed to him like it was not doing anything. The goal here would be to check if it does something or not with the E90. I guess it is supposed to start/stop recording a video on the SD card of the camera, isn't it?

Sushma7785, 2018-06-13:
Okay. I thought it was something to do with live video stream. 

> I guess it is supposed to start/stop recording a video on the SD card of the camera, isn't it?
Yes. That's right. I will try to check if it does something or not.

JonasVautherin, 2018-06-14:
Great, thanks a lot!

julianoes, 2018-06-14:
It could be that the mode was not set to video and so it would not work.

Sushma7785, 2018-06-16:
Start/stop capturing video on E90 is working.

JonasVautherin, 2018-06-16:
> Start/stop capturing video on E90 is working.

Thanks for testing! Can you explain what it is doing exactly? Is it creating a video file on the SD card?

> Mode was not the problem. Just that there was no implementation in the sample app for these functions, to test.

No actually @douglaswsilva and @byuarus had the right SDK version, but they did not really know what to expect with those functions. So I guess it was not clear if it was doing anything at all?

Sushma7785, 2018-06-17:
So startVideo() and stopVideo() in the Camera.swift, starts and stops video recording in the camera and stores it in the SD card.

Note: Camera mode should be in video mode to successfully record video.

JonasVautherin, 2018-06-17:
> Note: Camera mode should be in video mode to successfully record video.

Do you know what happens if it is not? Does it return an error, or does it just ignore it?

Sushma7785, 2018-06-18:
> Do you know what happens if it is not? Does it return an error, or does it just ignore it?

When the mode is not video, then it returns error "The operation couldn’t be completed". 
But one issue is, setting camera mode to video is not working, it always returns error. So I set video mode using our other app and then tried recording the video.

JonasVautherin, 2018-06-19:
Thanks Sushma, I'll have a look at `setMode` ASAP.