mavlink/MAVSDK-Swift #60: Add camera features to Sample app

JonasVautherin, 2018-06-15:
Not all the features are implemented yet, but many of them are (like `takePhoto`, etc), as detailed in #5.

Would be nice to add them/some of them in the app already.

JonasVautherin, 2018-06-15:
@ayameMBS Do you think you could have a look at this? So that the sample app would already have the structure to add the other features when they get implemented :-).

Sushma7785, 2018-06-18:
@JonasVautherin I can start working on this if that is okay.

JonasVautherin, 2018-06-19:
Thanks Sushma. Let's wait from a confirmation from @ayameMBS, maybe she already started that locally.

ayameMBS, 2018-06-20:
Sorry for the late reply. No, I haven't start yet that implementation. So @Sushma7785  you can do it.

JonasVautherin, 2019-02-08:
Not part of this repo anymore.

closed 2019-02-08 by JonasVautherin