mavlink/MAVSDK-Swift #72: Needs CI

mrpollo, 2018-08-31:
we need to guarantee a certain degree of safety on our SDK, we need to set up a recipe for building/running tests and host that on Jenkins.

- [ ] build script
- [ ] integration-tests script
- [ ] jenkins job
- [ ] jenkins <-> github integration
- [ ] readme build badge

JonasVautherin, 2018-08-31:
That's already started: @dagar set up Jenkins a few weeks ago already, and I've started looking into it. I just haven't had time to actually finish that, but it will come ASAP.

We will have everything from your list, except the integration tests. Integration tests are more tricky (Do we run one simulator instance per test? Does it scale? If we share one simulator, how do we make sure the drone is in the right state for the tests?).

But we will have the unit tests in bonus ;-).

mrpollo, 2018-08-31:
Awesome, let me know if there's anything you guys need.

re simulation: I would say starting a stopping SITL per test isn't that bad of an idea, at least it worked out for us in python

Example test [dronekit-python:dronekit/test/sitl/test_waypoints.py](https://github.com/dronekit/dronekit-python/blob/master/dronekit/test/sitl/test_waypoints.py)

SITL helper utility [dronekit-python:dronekit/test/__init.py__](https://github.com/dronekit/dronekit-python/blob/master/dronekit/test/__init__.py)

unipheas, 2019-11-14:
Is this still being worked on?

julianoes, 2019-11-14:
Yes I think some things are still missing.

JonasVautherin, 2020-11-27:
We now have github workflows.

closed 2020-11-27 by JonasVautherin