mavlink/MAVSDK-Swift #100: Arming Failed: Can't connect to 3DR Solo

hjames2, 2018-12-13:
Hello,

I am trying to get a working sample of the DronecodeSDK library working in my iOS app.  That being said, I am having trouble getting the SDK to actually connect to my 3DR Solo drone.  By following the basic guide in the README, I get the following error while attempting to arm the drone:

_Arming failed: The operation couldn't be completed. (SwiftGRPC.RPCError error 0)_

I have made sure that my iOS simulator is connected to the drone wifi.  I have also tried to run the sample project but to no avail (same error).

Any help would be appreciated.

Environment:
XCode -> 9.4.1
3DR Solo Drone
iOS Simulator (iPhone 8+)

JonasVautherin, 2018-12-13:
Hmm I've never tried the Solo. Does it get recognized? You would see some logs similar to:

```
[11:53:28|Info ] New device on: 127.0.0.1:14540 (udp_connection.cpp:200)
[11:53:28|Debug] Component Autopilot (1) added. (system_impl.cpp:369)
[11:53:28|Debug] Discovered 1 component(s) (UUID: 4294967298) (system_impl.cpp:515)
```

Would you mind trying with SITL first? If you have docker, you can try to run [this container](https://github.com/JonasVautherin/px4-gazebo-headless). I personally connect it directly to my phone (but it should work with the iOS simulator), like so:

    docker run --rm -it jonasvautherin/px4-gazebo-headless:v1.8.0 <ip of my iPhone>

julianoes, 2018-12-14:
@hjames2 what port does Solo send UDP MAVlink traffic on? I suspect the default QGC port 14550. By default the SDK connects on 14540. If you want to use 14550 instead you should be able to add that port like this:
```
add_udp_connection(14550);
```

hjames2, 2019-01-08:
Sorry, unfortunately other projects have pulled me away from this one and I have not had time to change and recompile the swiftSDK.  I will close this issue for now and possibly re-open it when/if I have time to test it further.  Thanks for your help thus far.

closed 2019-01-08 by hjames2