mavlink/MAVSDK-Swift #102: Carthage bootstrap timed out

caiyunfeide, 2018-12-19:
The Carthage import library used in the demo often times out of time. Can we download the library separately and introduce it into the Demo?

JonasVautherin, 2018-12-19:
We do not provide the binary files for download, and currently rely on Carthage for distribution.

However, the project is open source, which means that you can build the libraries from sources yourself.

caiyunfeide, 2018-12-20:
Can an objective-c project use CocoaPods and Carthage at the same time

JonasVautherin, 2018-12-20:
I guess, yes. Carthage just pulls sources and builds binaries out of them. If you can directly give binaries to Cocoapods (and I believe you can), then that should be possible.

caiyunfeide, 2018-12-20:
Thank you. I'll try again

caiyunfeide, 2018-12-20:
Excuse me, when I used Carthage bootstrap --platform ios, an error was thrown and the Build Failed
Task failed with exit code 65:
The/usr/bin/xcrun xcodebuild - project/Users/Miao/Desktop/DronecodeSDK - Swift - 0.2.1 / Carthage Checkouts/GRPC - Swift/SwiftGRPC - Carthage. Xcodeproj - scheme SwiftGRPC - Carthage - Package - the configuration Release - derivedDataPath/Users/Miao/Library/Caches/org. Carthage. CarthageKit/DerivedData / 10.1 _10B61 / GRPC - swift / 23 a0ebdee9613f615f2f2469ed3e700df5856417 - SDK iphoneos ONLY_ACTIVE_ARCH = NO CODE_SIGNING_REQUIRED = NO CODE_SIGN_IDENTITY = CARTHAGE = YES archive - 57 / nm4kbkpx12db53xg_3h71n1c0000gn archivePath/var/folders / / T/GRPC - swift SKIP_INSTALL = YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS = NO CLANG_ENABLE_CODE_COVERAGE = NO STRIP_INSTALLED_PRODUCT = NO (launched in/Users/Miao/Desktop/DronecodeSDK - Swift - 0.2.1 / Carthage/Checkouts/GRPC - Swift)

This usually are that the project itself failed to compile. Do check the xcodebuild log for more details: / var/folders / 57 / nm4kbkpx12db53xg_3h71n1c0000gn/T/Carthage - xcodebuild. XfqAum. Log
How can I solve this problem

julianoes, 2018-12-20:
It says "check these logs for more details". It would be good if you could paste these.
And please put anything you paste in-between ```.

JonasVautherin, 2018-12-20:
So it doesn't time out, but the build actually fails. And it seems to come from SwiftGRPC.

@caiyunfeide: Can you confirm that you use Xcode 10? SwiftGRPC doesn't build with Carthage on Xcode 10 yet, we are using Xcode 9.4.1 here.

caiyunfeide, 2018-12-20:
Yes, I'm using Xcode10..

caiyunfeide, 2018-12-20:
Where can I find the source code of backend. Framework

julianoes, 2018-12-20:
The backend is built manually from the C++ Dronecode SDK.
Check the build instructions "Building the backend"  here:
https://sdk.dronecode.org/en/contributing/build.html

JonasVautherin, 2019-01-08:
Closing here as it is the same issue as #104.

closed 2019-01-08 by JonasVautherin