mavlink/MAVSDK-Swift #114: Cannot build GRPC-Swift package in SwiftGRPC-Package.xcodeproj with Carthage

Renosi, 2019-03-22:
Currently I'm trying to use the Carthage build to add the frameworks. I am running Carthage version 0.32 and Swift 4.2 

I am having issues when trying to run 

``carthage bootstrap --platform ios``

I am currently stuck with the build error: 

``Downloading RxSwift.framework binary at "Atomic.2``
``*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj``
``Build Failed``
	``Task failed with exit code 65:``
	``/usr/bin/xcrun xcodebuild -project /Users/renosimomoh/Drone\ Delivery\ /Medical\ ``
``Drone/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath `` ``/Users/renosimomoh/Library/Caches/org.carthage.CarthageKit/DerivedData/10.1_10B61/grpc`` ``-swift/b7659e1031b441da2db60fa879e8157799730f02 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath`` ``/var/folders/_s/vpqkvyq96rl9q9kj_b50w1s80000gn/T/grpc-swift SKIP_INSTALL=YES`` ``GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO`` ``STRIP_INSTALLED_PRODUCT=NO (launched in /Users/renosimomoh/Drone Delivery /Medical`` `` Drone/Carthage/Checkouts/grpc-swift)``

``This usually indicates that project itself failed to compile. Please check the xcodebuild log`` ``for more details: /var/folders/_s/vpqkvyq96rl9q9kj_b50w1s80000gn/T/carthage ``xcodebuild.aT7PEJ.log``

julianoes, 2019-03-25:
What's in the logfile /var/folders/_s/vpqkvyq96rl9q9kj_b50w1s80000gn/T/carthage xcodebuild.aT7PEJ.log ?

Renosi, 2019-03-25:
[carthage-xcodebuild.rMztLb.log](https://github.com/Dronecode/DronecodeSDK-Swift/files/3002730/carthage-xcodebuild.rMztLb.log)

julianoes, 2019-03-25:
Ok thanks. I see this but I'm not sure what's wrong.

```
/var/folders/_s/vpqkvyq96rl9q9kj_b50w1s80000gn/T/SchemeScriptAction-gTAIKj: line 2: cd: /Users/renosimomoh/Library/Mobile: No such file or directory
2019-03-25 10:10:55.498 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
2019-03-25 10:10:56.219 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
2019-03-25 10:10:56.296 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
2019-03-25 10:10:56.306 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
2019-03-25 10:10:56.317 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
2019-03-25 10:10:56.330 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
2019-03-25 10:10:56.343 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
2019-03-25 10:10:56.356 xcodebuild[1995:153564] [MT] DVTDeveloperPaths: WARNING: Environment variable XCODE_DEVELOPER_DIR_PATH is set, but is not an absolute path. The variable will be ignored. ()
error: root manifest not found
ruby: No such file or directory -- fix-carthage-paths.rb (LoadError)
```

Can you check that you have

julianoes, 2019-03-25:
What's the content of your `Cartfile`?

Btw. you can put ` (1 tick) for inline or ``` (3 ticks) for multiline around your code

julianoes, 2019-03-25:
It builds for me with this output:
```
carthage bootstrap --platform ios
*** Checking out DronecodeSDK-Swift at "0.3.0"
*** Checking out grpc-swift at "b7659e1031b441da2db60fa879e8157799730f02"
*** Checking out RxSwift at "4.4.0"
*** xcodebuild output can be found in /var/folders/1y/9hm0t7pj055996vnd398ns6r0000gn/T/carthage-xcodebuild.mwry9J.log
*** Downloading binary-only framework backend at "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json"
*** Downloading binary-only framework MobileVLCKit at "https://code.videolan.org/videolan/VLCKit/raw/master/Packaging/MobileVLCKit.json"
*** Downloading RxSwift.framework binary at "Atomic"
*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj
*** Building scheme "Dronecode_SDK_Swift" in Dronecode-SDK-Swift.xcodeproj
```

That's also with carthage 0.32.0, and XCode 10.1, and Swift version 4.2.1:

```
carthage version
0.32.0

xcrun swift -version
Apple Swift version 4.2.1 (swiftlang-1000.11.42 clang-1000.11.45.1)
Target: x86_64-apple-darwin18.2.0
```

Renosi, 2019-03-25:
Thanks for looking into this and yeah I didn't know about the multiline. 

My cartfile contains 
`github "Dronecode/DronecodeSDK-Swift" ~> 0.3.0`

JonasVautherin, 2019-03-25:
> ruby: No such file or directory -- fix-carthage-paths.rb (LoadError)

Do you have ruby installed (not sure if it comes by default on macOS or not)? `$ which ruby` or `$ ruby --version`?

Renosi, 2019-03-25:
I definitely have Ruby installed 

```MacBook-Pro-2:Medical Drone GC renosimomoh$ ruby -v```
```ruby 2.3.7p456 (2018-03-28 revision 63024) [universal.x86_64-darwin18]```

julianoes, 2019-03-25:
I just removed my `Cartfile.resolved` and used `github "Dronecode/DronecodeSDK-Swift" ~> 0.3.0` in the `Cartfile` and it worked fine:

I suggest you clean everything that Carthage has built already:
```
cd where/youre/building
rm -rf Carthage
```

as well as the Carthage caches:
```
rm -rf ~/Library/Caches/org.carthage.CarthageKit ~/Library/Caches/carthage
```

Renosi, 2019-03-25:
I tried everything but I still get the same error. I updated Ruby to the latest version, I removed the `Cartfile` and `Cartfile.resolved` as well as cleaning everything built already up to and including Carthage caches.

JonasVautherin, 2019-03-25:
> cd: /Users/renosimomoh/Library/Mobile: No such file or directory

What's `/Users/renosimomoh/Library/Mobile`?

davidknezic, 2019-03-25:
@Renosi I've noticed this line at the end of your log file:

```
error: Build input files cannot be found: '/Users/renosimomoh/Library/Mobile Documents/com~apple~CloudDocs/4th Year Files/project /Medical Drone GC/Carthage/Checkouts/grpc-swift/.build/checkouts/swift-protobuf.git-3495140620946553999/Sources/SwiftProtobuf/source_context.pb.swift'
```

Can you check, does the folder `Carthage/Checkouts/grpc-swift/.build/checkouts/swift-protobuf.git-3495140620946553999/` in your project exist? I've had a similar problem and thought that this was resolved with https://github.com/grpc/grpc-swift/pull/360.

Renosi, 2019-03-25:
> /Users/renosimomoh/Library/Mobile

Not sure where you saw that but it should say `/Users/renosimomoh/Library/Mobile Documents`
It's basically just the directory to my main iCloud Drive folder.

JonasVautherin, 2019-03-25:
That is an error in your log. Can you try to copy your cartfile in a tmp directory without spaces (e.g. `/tmp/trying-to-build/Cartfile`), with the same content as you have:

```
github "Dronecode/DronecodeSDK-Swift" ~> 0.3.0
```

And run `$ carthage bootstrap --platform ios` from there?

Renosi, 2019-03-25:
Just tried to build it in the `/tmp` folder. Seemingly the exact same error but I have attached the log files in case you wanted to have a look 
[carthage-xcodebuild.tZfHOU.log](https://github.com/Dronecode/DronecodeSDK-Swift/files/3004776/carthage-xcodebuild.tZfHOU.log)

julianoes, 2019-03-25:
@Renosi oh can you please try this:

```
gem install --user xcodeproj
```

Renosi, 2019-03-25:
I tried that and got this at my output at the terminal but I'm not quite sure how this solves the problem. 

```MacBook-Pro-2:tmp renosimomoh$ gem install --user xcodeproj
WARNING:  You don't have /Users/renosimomoh/.gem/ruby/2.4.0/bin in your PATH,
	  gem executables will not run.
Successfully installed atomos-0.1.3
Successfully installed CFPropertyList-3.0.0
Successfully installed claide-1.0.2
Successfully installed colored2-3.1.2
Successfully installed nanaimo-0.2.6
GemWrappers: Can not wrap missing file: xcodeproj
Successfully installed xcodeproj-1.8.1
Parsing documentation for atomos-0.1.3
Parsing documentation for CFPropertyList-3.0.0
Parsing documentation for claide-1.0.2
Parsing documentation for colored2-3.1.2
Parsing documentation for nanaimo-0.2.6
Parsing documentation for xcodeproj-1.8.1
Done installing documentation for atomos, CFPropertyList, claide, colored2, nanaimo, xcodeproj after 2 seconds
6 gems installed
```

julianoes, 2019-03-25:
Try `which xcodeproj` to see if the command works. If not, do this `export PATH=$PATH:/Users/renosimomoh/.gem/ruby/2.4.0/bin` to temporary add it to the path.

Renosi, 2019-03-25:
When I try the command `which xcodeproj`. I get`/Users/renosimomoh/.gem/ruby/2.4.0/bin/xcodeproj`
so I know the path has been added but still getting the same error

JonasVautherin, 2019-03-25:
> Seemingly the exact same error

I don't think it is the same error. Let's try to fix the `/tmp` build just in case, and we'll come back to the "folder name with space" later.

>  so I know the path has been added but I'm still getting the same error

Would you mind pasting the latest log of that? The previous one was saying:

> /Users/renosimomoh/.rvm/rubies/ruby-2.4.2/lib/ruby/site_ruby/2.4.0/rubygems/core_ext/kernel_require.rb:54:in `require': cannot load such file -- xcodeproj (LoadError)

which sounded like `xcodeproj` was not installed (and it apparently wasn't installed before, given that your output said "Successfully installed xcodeproj-1.8.1").

Renosi, 2019-03-26:
I wasn't sure If I ran`gem install -user xcodeproj`so I ran it again this morning to make sure. 
[carthage-xcodebuild.lfzqxj.log](https://github.com/Dronecode/DronecodeSDK-Swift/files/3007132/carthage-xcodebuild.lfzqxj.log)

JonasVautherin, 2019-03-26:
> /var/folders/_s/vpqkvyq96rl9q9kj_b50w1s80000gn/T/SchemeScriptAction-TV17C1: line 2: cd: /tmp/Proj: No such file or directory

Do you have a space in your name again? What is `/tmp/Proj`?

Renosi, 2019-03-26:
I've run the folder without the spaces to a name and I still got the error with the build input files not being found in the folder. 
[carthage-xcodebuild.bqvVqw.log](https://github.com/Dronecode/DronecodeSDK-Swift/files/3008002/carthage-xcodebuild.bqvVqw.log)

JonasVautherin, 2019-03-26:
```
/Users/renosimomoh/.rvm/rubies/ruby-2.4.2/lib/ruby/site_ruby/2.4.0/rubygems/core_ext/kernel_require.rb:54:in `require': cannot load such file -- xcodeproj (LoadError)
	from /Users/renosimomoh/.rvm/rubies/ruby-2.4.2/lib/ruby/site_ruby/2.4.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	from fix-carthage-paths.rb:1:in `<main>'
```

I believe it is failing at [this line](https://github.com/grpc/grpc-swift/blob/master/fix-carthage-paths.rb#L1), which is `require xcodeproj`. It is also the same error as e.g. [here](https://github.com/CocoaPods/pod-template/issues/222).

Would you mind trying to make a minimal ruby script (say `test.rb`) containing `require xcodeproj` and trying to run it?

Renosi, 2019-03-27:
> ```
> /Users/renosimomoh/.rvm/rubies/ruby-2.4.2/lib/ruby/site_ruby/2.4.0/rubygems/core_ext/kernel_require.rb:54:in `require': cannot load such file -- xcodeproj (LoadError)
> 	from [...]

What are the prerequisites for running the script? I tried to run the testy.rb file with `require xcodeproj` in the Dronecode SDK swift folder and I got:

``DronecodeSDK-Swift-master renosimomoh$ ruby testy.rb
testy.rb:1:in `<main>': undefined local variable or method `xcodeproj' for main:Object (NameError)
``

julianoes, 2019-03-27:
I get the same output with the test file, but like this it works:
```
require 'xcodeproj'
```

Renosi, 2019-03-27:
Ok now I'm getting this 

``todors-iphone:DronecodeSDK-Swift-master renosimomoh$ ruby testy.rb
/Users/renosimomoh/.rvm/rubies/ruby-2.4.2/lib/ruby/site_ruby/2.4.0/rubygems/core_ext/kernel_require.rb:54:in `require': cannot load such file -- xcodeproj (LoadError)
	from /Users/renosimomoh/.rvm/rubies/ruby-2.4.2/lib/ruby/site_ruby/2.4.0/rubygems/core_ext/kernel_require.rb:54:in `require'
	from testy.rb:1:in `<main>'``

JonasVautherin, 2019-03-27:
Which seems similar to [the error in your log](https://github.com/Dronecode/DronecodeSDK-Swift/issues/114#issuecomment-476628054).

Seems like your issue is that `xcodeproj` is not installed properly. Running the `test.rb` script @julianoes shows above works both for him and for me.

Renosi, 2019-03-27:
Yes you're right. I logged into a guest account on the same PC and ensured that the ruby path was added properly by doing it several times. I think the issue was that the Ruby path I was adding to the file was 2.4.0 when the version was actually 2.3.0 meaning that not all of the gems were properly installed. It works now.

julianoes, 2019-03-27:
Great, thx for the feedback.

PR #115 (2019-03-27): README: add xcodeproj dependency
Presumably fixes #114 .
closed 2019-03-27 by julianoes