mavlink/MAVSDK-Swift #120: Error building SDK from Carthage

douglaswsilva, 2019-05-06:
Get this error when trying to run `carthage bootstrap --platform ios`:

```
```

douglaswsilva, 2019-05-06:
[douglog.log](https://github.com/Dronecode/DronecodeSDK-Swift/files/3149893/douglog.log)

JonasVautherin, 2019-05-06:
Seems like they released [gRPC-Swift 0.9.0](https://github.com/grpc/grpc-swift/releases/tag/0.9.0) with "fixes for Swift 5".

@douglaswsilva: would you mind [trying gRPC-Swift 0.9.0](https://github.com/Dronecode/DronecodeSDK-Swift/blob/master/Cartfile#L1)? Should be a matter of editing the SDK Cartfile and running Carthage again.

byuarus, 2019-05-07:
@JonasVautherin I've just checked and I see that [gRPC-Swift 0.9.0](https://github.com/grpc/grpc-swift/releases/tag/0.9.0) doesn't work with the XCode 10.1 (Swift 4.2) but [the latest commit in master](https://github.com/grpc/grpc-swift/commit/67a5352f179751f6aef4e709c030308722f97073) works.

JonasVautherin, 2019-05-07:
Oh, good find!

Just to recap then: it broke, but...
* We haven't changed the release of gRPC-Swift that we use
* We are still building with Swift 4.2

What could have changed that would break the build? Could it be a change in RxSwift? Like instead of `~> 4.0` should we fix it to some version?

byuarus, 2019-05-07:
Hi @JonasVautherin, 
- I think that the Carthage was broken for a while when you have changed the version of gRPC-Swift that you use. I think it was the only issue for the Carthage builds. Please check my PR #122  that should fix Carthage support for the SDK. 
- Yes, we're still using XCode 10.1 and Swift 4.2. We can't migrate our code to Swift 5 and XCode 10.2 yet as some of our 3rd party libraries haven't done the migration.

PR #123 (2019-05-09): grpc-swift 0.8.2 apparently broke the build.
Using `~>` for Carthage dependencies was a mistake, and that broke our build. Let's stick to `==` now.

Resolves #120, replaces #122.

@douglaswsilva, @byuarus: FYI
closed 2019-05-09 by JonasVautherin