mavlink/MAVSDK-Swift #125: Error with backend.json 0.16.0

unipheas, 2019-05-10:
When performing the command 
`Carthage build --platform iOS --no-skip-current --verbose` 
I am receiving the following error 
`*** Downloading binary-only framework backend at "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json"
No available version for binary "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json" satisfies the requirement: == 0.16.0`
I had to change the Cartfile to 0.15.0 for it to work.

*** I am using Shadowsocks and proxying through the GFW in China, if this has anything to do with it. However, 0.15.0 worked so I'm not sure about that.

JonasVautherin, 2019-05-10:
I had it as well, I believe. Can you try to [clean Carthage cache](https://stackoverflow.com/questions/46346743/carthage-error-no-available-version-for-binary)? Seems like Carthage sometimes needs that when using binary-only frameworks :roll_eyes:.

    rm -rf ~/Library/Caches/org.carthage.CarthageKit/
    rm -rf ~/Library/Caches/carthage/

unipheas, 2019-05-13:
This was resolved. Thank you!

closed 2019-05-13 by unipheas