mavlink/MAVSDK-Swift #128: Task failed with exit code 65

unipheas, 2019-05-27:
I created a new Xcode Project for iOS. It's just a default app, nothing has changed once the create button was clicked. I then went into the terminal and into the project directory and typed the following commands as per the instructions for this repo.
```
gem install --user xcodeproj
touch Cartfile
vi Cartfile
# add github "Dronecode/DronecodeSDK-Swift" ~> 0.3.0
carthage bootstrap --platform ios
```
I then receive this error, which was also mentioned [here](https://github.com/Dronecode/DronecodeSDK-Swift/issues/120 ) but it was for the Swift SDK itself.
```
carthage bootstrap --platform ios
*** Checking out grpc-swift at "0.8.2"
*** Checking out DronecodeSDK-Swift at "0.4.1"
*** Checking out RxSwift at "4.5.0"
*** xcodebuild output can be found in /var/folders/gs/_gq_c3k97db9b1xg6dg84lww0000gn/T/carthage-xcodebuild.3zbbvi.log
*** Downloading binary-only framework backend at "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json"
*** Downloading RxSwift.framework binary at "[4,5]"
***  Skipped installing RxSwift.framework binary due to the error:
	"Incompatible Swift version - framework was built with 5.0 (swiftlang-1001.0.69.5 clang-1001.0.46.3) and the local version is 5.0.1 (swiftlang-1001.0.82.4 clang-1001.0.46.5)."

    Falling back to building from the source
*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj
Build Failed
	Task failed with exit code 65:
	/usr/bin/xcrun xcodebuild -project /Users/unipheas/Developer/px4/DroneCode_iOS/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/unipheas/Library/Caches/org.carthage.CarthageKit/DerivedData/10.2.1_10E1001/grpc-swift/0.8.2 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/gs/_gq_c3k97db9b1xg6dg84lww0000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO (launched in /Users/unipheas/Developer/px4/DroneCode_iOS/Carthage/Checkouts/grpc-swift)

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/gs/_gq_c3k97db9b1xg6dg84lww0000gn/T/carthage-xcodebuild.3zbbvi.log
```
The log has this to say
```
ld: warning: directory not found for option '-L/usr/local/homebrew/Cellar/nghttp2/1.35.1/lib'
ld: library not found for -lnghttp2
clang: error: linker command failed with exit code 1 (use -v to see invocation)

** ARCHIVE FAILED **

The following build commands failed:
	SwiftCodeGeneration normal armv7
	SwiftCodeGeneration normal arm64
	Ld /Users/unipheas/Library/Caches/org.carthage.CarthageKit/DerivedData/10.2.1_10E1001/grpc-swift/0.8.2/Build/Intermediates.noindex/ArchiveIntermediates/SwiftGRPC-Package/IntermediateBuildFilesPath/SwiftGRPC-Carthage.build/Release-iphoneos/BoringSSL.build/Objects-normal/armv7/BoringSSL normal armv7
(3 failures)
```
I looked into installing nghttp2 through Brew but it seems to already be installed. This seems to be slightly different from issue #120

JonasVautherin, 2019-05-27:
> *** Checking out grpc-swift at "0.8.2"

You should not get grpc-swift `0.8.2`, which apparently broke the iOS build. It was fixed by #123 in [Cartfile](https://github.com/Dronecode/DronecodeSDK-Swift/blob/master/Cartfile) and in [Cartfile.resolved](https://github.com/Dronecode/DronecodeSDK-Swift/blob/master/Cartfile.resolved#L3).

I see that you ran `carthage bootstrap --platform ios`, which is correct. So either you are not on latest `master`, or you may need to remove your `Carthage` directory and possibly the Carthage cache (see #125).

I know that this is a bit annoying, sorry about it :-/. I am not sure why you got `0.8.2` (maybe you ran `carthage update` instead of `carthage bootstrap` at some point?).

unipheas, 2019-05-27:
Yea, that may have been it. I'll reclone the repo and start from scratch and try again. Thank you!

unipheas, 2019-05-27:
Build was successful, Thank you!

closed 2019-05-27 by unipheas