mavlink/MAVSDK-Swift #132: "SchedulerType = ConcurrentDispatchQueueScheduler(qos: .background)" has crash

CoDancer, 2019-06-24:


unipheas, 2019-06-24:
@CoDancer Can you explain what's going on in more detail please? Try to include any error messages or the steps to reproduce what you have found.

CoDancer, 2019-06-24:
Thanks! There is  a project that I use the SDK, you can download and run the project, and you will see what I said.(The project url: https://github.com/CoDancer/MAVLinkSwiftDemo)

unipheas, 2019-06-24:
@CoDancer It would really help us out if you can explain what's going on and try to give us more information. Providing a project isn't the ideal way to let us know what the exact issue is. Here are some links that can help you formulate an issue.

https://wiredcraft.com/blog/how-we-write-our-github-issues/
https://upthemes.com/blog/2014/02/writing-useful-github-issues/

Thank you!

CoDancer, 2019-06-24:
@unipheas I just want to create an object of the drone in a method, so I use the "let drone = Drone.init(address: "", port: 8090)", but in the method of "public init(address: String,
                port: Int32,
                scheduler: SchedulerType = ConcurrentDispatchQueueScheduler(qos: .background))" has crash, 
[image]
like the image, and the debug is "Message from debugger: Terminated due to signal 9"

JonasVautherin, 2019-06-24:
@CoDancer: Which version of MAVSDK-Swift do you use? Are you getting the frameworks through Carthage?

CoDancer, 2019-06-24:
[image]

JonasVautherin, 2019-06-24:
Looks like your `Cartfile.resolved`, so I guess that's fine. I'll try to see if I can run your example. In the meantime, can you try our [example app](https://github.com/mavlink/MAVSDK-Swift-Example) and tell me if that fails as well?

unipheas, 2019-06-25:
@CoDancer I had a look at your project. I can tell you that the reason it's crashing is because you do not have your storyboard view connected to your view controller class. That said, it looks like you're not even using it. So, unfortunately, it has nothing to do with the SDK itself but your app itself. I would suggest going through http://hackingwithswift.com/100 to get a better understanding of the iOS development environment.

closed 2019-06-25 by unipheas

CoDancer, 2019-06-25:
@unipheas but I run my project if I shield the "Drone.init(address: "", port: 8090)",my project is normal.
like this.
[image]

unipheas, 2019-06-25:
@CoDancer I'm playing around with your project. In the meantime can you please follow these [instructions](https://gist.github.com/unipheas/a5f94751c665995c56fba3c3af61a7a0#swift-available-for-macos).

CoDancer, 2019-06-25:
@unipheas Do you mean this? Installation the QGroundControl on Mac.
[image]
If so：
I had installed it.
[image]

unipheas, 2019-06-25:
@CoDancer No. The Swift instructions at the bottom of that page, where the link directs too.

unipheas, 2019-06-25:
@CoDancer By the way. Are you in China and do you have WeChat? I can speak with you directly on this if you'd like.

CoDancer, 2019-06-25:
@unipheas  Ok, please add my Wechat,1848225396

douglaswsilva, 2019-06-25:
I managed to get your project running. The reason for the crash I found was that you imported the libraries manually but didn't link them properly. To fix it, you can either add them to the **Embedded Binaries** or add a script to copy them during the build process in the **Build Phases**.

douglaswsilva, 2019-06-25:
That being said, I don't think you need **Drone** folder in your project, since everything is already included in the Dronecode_SDK_Swift.framework.
Also, I know you did it for testing purposes, but I wouldn't call the actions inside app delegate, try to move that logic somewhere else. 

Good luck!