mavlink/MAVSDK-Swift #136: Camera setSetting not working

douglaswsilva, 2019-07-11:
When trying to set any settings for E90 camera I get this error:

```
[06:18:07|Error] Error: set param busy timeout: CAM_COLORMODE (mavlink_parameters.cpp:470)
failure: setSetting() CameraError(code: Dronecode_SDK_Swift.Camera.CameraResult.Result.error, description: "Error")
```

JonasVautherin, 2019-07-11:
Thanks for reporting! Is that new (I think it used to work)? And if yes, does it still work on the older version?

Just trying to see if that's coming from MAVSDK or maybe an update in the E90...

douglaswsilva, 2019-07-11:
@JonasVautherin Yes, I had reported this issue a while ago and it seemed to have gotten fixed by a camera firmware update. But with another update, it got broken again. I was wondering if there's anything we can do on the SDK side so this functionality won't rely on camera firmware being changed. What do you think?

JonasVautherin, 2019-07-12:
The [MAVLink testing suite](https://github.com/Auterion/mavlink-testing-suite/blob/master/src/tests/camera_settings.h) actually does test that at the C++ level. If that is an issue in the firmware, it should be flagged by this test.

Ideally, Yuneec could run that against the E90 before releasing new firmwares, I guess... @julianoes is that correct?

@douglaswsilva: could you try to run those tests against your H520?

julianoes, 2019-07-12:
Hm I'm sorry. I need to go back and check when I changed the camera plugin last. Presumably that's when I messed it up.

julianoes, 2019-07-12:
The other thing is that we cache the xml files for the E90, E50, ET and if the camera changes we need to get the updates for MAVSDK as well.

byuarus, 2019-07-15:
@julianoes Do we need to update the MAVSDK version every time Yuneec release a new version of camera firmware with new xml file?

JonasVautherin, 2019-07-17:
I guess that ideally, MAVSDK would download that file from the camera. We could possibly do that once the [MAVLink FTP plugin from Matej](https://github.com/mavlink/MAVSDK/pull/774) is in. But then Yuneec should expose that file over MAVLink FTP as well.

@julianoes is that correct?

julianoes, 2019-07-18:
> @julianoes Do we need to update the MAVSDK version every time Yuneec release a new version of camera firmware with new xml file?

Yes because the file is cached in the SDK. If Yuneec implements the MAVLink FTP server in the camera we could download it via MAVLink FTP as @JonasVautherin said.

Alternatively, if the bridge between the ST10C and iOS supported some sort of IP networking, we could access the http server of the camera in MAVSDK. Right now we can't because the bridge only forwards MAVLink (and RC I think).

douglaswsilva, 2019-09-04:
Seems to be resolved by drone team once again.

closed 2019-09-04 by douglaswsilva