mavlink/MAVSDK-Swift #137: CameraPhotoIntervalS not working in MissionItem

douglaswsilva, 2019-07-11:
Drone will take picture every second despite of the value we set. For instance:

```
Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: location.latitude,
                                         longitudeDeg: location.longitude,
                                         relativeAltitudeM: 4.0,
                                         speedMS: 4.0,
                                         isFlyThrough: false,
                                         gimbalPitchDeg: -30,
                                         gimbalYawDeg: 0,
                                         cameraAction: .startPhotoInterval, 
                                         loiterTimeS: 0,
                                         cameraPhotoIntervalS: 5)

```

julianoes, 2019-07-12:
Funny, that's a feature, right? :smile: 

Do you think that's a regression with MAVSDK?

douglaswsilva, 2019-07-15:
🙃
I don't remember when it stoped working though, @byuarus might remember.

byuarus, 2019-07-15:
@julianoes I'm not sure when it broke. It was working with the first Dronecode SDK release that had this feature and Yuneec FW at that time. 
@julianoes Do you think it could be an incompatibility with the autopilot/camera firmware version or this issue is on the MAVSDK side only?

JonasVautherin, 2019-07-17:
Thinking about it, one thing you can try is this:

1. Upload the mission to the drone
2. Download the mission from QGC
3. Check in QGC if the time interval is set correctly. If yes, I would assume that MAVSDK did its job correctly. If not, MAVSDK possibly messed up somewhere... :smile:

byuarus, 2019-07-17:
@JonasVautherin  
I did the test and it looks that the issue is on the MAVSDK side.

My test:
1) I set `cameraPhotoIntervalS` to 4 sec for every `MissionItem` from our app
2) Uploaded mission to the drone from our app
3) Downloaded the mission from QGC and `Start image capture` shows interval 1 sec 
4) Did a flight and camera tried to take photos with 1-sec interval
5) Corrected the mission in QGC, changed `Start image capture` interval to 4 sec
6) Uploaded the mission from QGC
7) Did a flight and camera took photos with 4-sec interval as expected

JonasVautherin, 2019-07-17:
> It was working with the first Dronecode SDK release that had this feature and Yuneec FW at that time.

Are we sure of that? I believe this field was always ignored actually (see fix [here](https://github.com/mavlink/MAVSDK/pull/801)). Could it be that you were starting the timelapse differently before (e.g. with the camera plugin)?

byuarus, 2019-07-17:
> Could it be that you were starting the timelapse differently before (e.g. with the camera plugin)?
Yes, it could be the case.

byuarus, 2019-07-17:
@JonasVautherin I've tested your [fix](https://github.com/mavlink/MAVSDK/pull/801) and it worked 👍 
Thank you!

closed 2019-07-18 by julianoes