mavlink/MAVSDK-Swift #138: Command ack causing connection issues

douglaswsilva, 2019-07-11:
After a lot of searching I found out one thing that has been causing a lot of connection issues between drone and sdk. 

Reproducible steps:
1. Use all the latest backend and swift SDK in DronecodeSDK_Swift_Example and connect to vehicle via WIFI while connected to debugger via XCode (I tested with H520-G and E90 from Yuneec)
2. Subscribe to cameraStatus:
```
        _ = drone.camera.cameraStatus
            .subscribe(onNext: { status in
                NSLog("Camera Status: \(status)")
            }, onError: { error in
                NSLog("Error cameraStatusSubscription: \(error.localizedDescription)")
            })
```
3. Try to set some camera setting:
```
        let option =  Camera.Option(optionID: "1", optionDescription: "")
        let setting = Camera.Setting(settingID: "CAM_COLORMODE", settingDescription: "", option: option)
        _ = drone.camera.setSetting(setting: setting)
            .do(onError: { error in
                NSLog("failure: setSetting() \(error)")
            }, onCompleted: {
                NSLog("Completed")
            })
            .subscribe()
``` 
4. You will see something like this in the debugger:
```
[06:37:23|Debug] Lost 3762002188609060911 (mavsdk_impl.cpp:360)
Drone disconnected
[06:37:23|Debug] Discovered 3 component(s) (UUID: 3762002188609060911) (system_impl.cpp:563)
Drone connected with UUID : 3762002188609060911
[06:37:23|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:219)
[06:37:24|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:219)
[06:37:24|Warn ] sending again, retries to do: 1  (527). (mavlink_commands.cpp:219)
[06:37:25|Error] Retrying failed (527) (mavlink_commands.cpp:235)
[06:37:25|Warn ] Command ack 527 not matching our current command: 525 (mavlink_commands.cpp:141)
[06:37:25|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:219)
[06:37:26|Warn ] sending again, retries to do: 2  (525). (mavlink_commands.cpp:219)
[06:37:26|Warn ] sending again, retries to do: 1  (525). (mavlink_commands.cpp:219)
[06:37:27|Error] Retrying failed (525) (mavlink_commands.cpp:235)
[06:37:27|Warn ] sending again, retries to do: 3  (520). (mavlink_commands.cpp:219)
[06:37:28|Warn ] sending again, retries to do: 2  (520). (mavlink_commands.cpp:219)
[06:37:28|Warn ] Command ack 527 not matching our current command: 520 (mavlink_commands.cpp:141)
[06:37:28|Warn ] sending again, retries to do: 1  (520). (mavlink_commands.cpp:219)
[06:37:29|Error] Retrying failed (520) (mavlink_commands.cpp:235)
[06:37:29|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:219)
[06:37:30|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:219)
[06:37:30|Warn ] sending again, retries to do: 1  (527). (mavlink_commands.cpp:219)
[06:37:31|Error] Retrying failed (527) (mavlink_commands.cpp:235)
[06:37:31|Warn ] Command ack 527 not matching our current command: 525 (mavlink_commands.cpp:141)
[06:37:31|Warn ] Command ack 525 not matching our current command: 520 (mavlink_commands.cpp:141)
[06:37:31|Warn ] sending again, retries to do: 3  (520). (mavlink_commands.cpp:219)
[06:37:32|Warn ] sending again, retries to do: 2  (520). (mavlink_commands.cpp:219)
[06:37:32|Warn ] sending again, retries to do: 1  (520). (mavlink_commands.cpp:219)
[06:37:33|Error] Retrying failed (520) (mavlink_commands.cpp:235)
[06:37:33|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:219)
[06:37:34|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:219)
[06:37:34|Warn ] Command ack 525 not matching our current command: 527 (mavlink_commands.cpp:141)
[06:37:34|Warn ] sending again, retries to do: 1  (527). (mavlink_commands.cpp:219)
[06:37:35|Info ] heartbeats timed out (system_impl.cpp:312)
[06:37:35|Debug] Lost 3762002188609060911 (mavsdk_impl.cpp:360)
Drone disconnected
[06:37:35|Debug] Discovered 3 component(s) (UUID: 3762002188609060911) (system_impl.cpp:563)
Drone connected with UUID : 3762002188609060911
[06:37:35|Error] Retrying failed (527) (mavlink_commands.cpp:235)
[06:37:35|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:219)
[06:37:36|Warn ] sending again, retries to do: 2  (525). (mavlink_commands.cpp:219)
```

After that you won't be able to send any command to the drone and all your telemetry will be frozen. Only way to fix it is to reset both app and drone.

unipheas, 2019-07-11:
@douglaswsilva is this a continuation of #136 and #137?

douglaswsilva, 2019-07-11:
It is triggered by #136 but in this case, it is `cameraStatus` subscription that actually causes the issue of locking communication with the drone.

douglaswsilva, 2019-07-11:
@julianoes Yuneec people asked if you could take a look at this and let them know if there's anything they can do on their side.

julianoes, 2019-07-11:
I have fixed that in the past, it's probably some deadlock. Can you reproduce it with the Swift Example? Or the Python or C++ code? If yes, I'll be quicker in debugging it.

douglaswsilva, 2019-07-11:
Yes, I can reproduce in the Swift Example app (https://github.com/mavlink/MAVSDK-Swift-Example) using the latest master of the Swift SDK. You just need to add the two pieces of code I posted above. First connect to drone via wifi, then subscribe to camera Status, and last try to set some camera setting and you should see the issue. 

If you can't reproduce it and think it is a firmware issue, if you don't mind asking Dario or Joe to take a look at it as well?

douglaswsilva, 2019-07-11:
@julianoes Do you want me to try to reproduce it in python. I don't have the python build system installed, but can give it a shot. It will just take a while to get everything set up.

unipheas, 2019-07-12:
@douglaswsilva I wrote some instructions that might help you speed the process along. 
https://gist.github.com/unipheas/a5f94751c665995c56fba3c3af61a7a0

julianoes, 2019-07-12:
@douglaswsilva if it's reproducible in the Swift example that's fine. I'll try that next week.

douglaswsilva, 2019-07-13:
Thank you @unipheas, that's actually a really nice tutorial!

julianoes, 2019-07-13:
I pushed some fixes that we can try: https://github.com/mavlink/MAVSDK/pull/796
I haven't been able to test them yet or reproduce the issue though.

JonasVautherin, 2019-07-16:
FYI: I'll merge the PR as soon as the CI passes, and I'll make a new release of MAVSDK for you guys to test.

douglaswsilva, 2019-07-16:
Hi guys, 
I just tested the new `mavsdk_server` that @byuarus built from https://github.com/mavlink/MAVSDK/pull/796 and unfortunately we are still getting the same issue. I tested using the `DronecodeSDK_Swift_Example` app connected to the E90 via wifi (same thing as I posted above). I subscribe to `cameraStatus` and then try to `setSetting` after, this is the output (I was printing the callback from camera status, so it was printing values it until I call setSetting): 

```
Camera Status: CameraStatus(videoOn: false, photoIntervalOn: false, usedStorageMib: 15342.5, availableStorageMib: 15082.0, totalStorageMib: 30424.5, recordingTimeS: 0.0, mediaFolderName: "727E90HD", storageStatus: Dronecode_SDK_Swift.Camera.CameraStatus.StorageStatus.formatted)
2019-07-16 11:49:59.455181-0700 DronecodeSDK_Swift_Example[879:207596] Camera Status: CameraStatus(videoOn: false, photoIntervalOn: false, usedStorageMib: 15342.5, availableStorageMib: 15082.0, totalStorageMib: 30424.5, recordingTimeS: 0.0, mediaFolderName: "727E90HD", storageStatus: Dronecode_SDK_Swift.Camera.CameraStatus.StorageStatus.formatted)
Camera Status: CameraStatus(videoOn: false, photoIntervalOn: false, usedStorageMib: 15342.5, availableStorageMib: 15082.0, totalStorageMib: 30424.5, recordingTimeS: 0.0, mediaFolderName: "727E90HD", storageStatus: Dronecode_SDK_Swift.Camera.CameraStatus.StorageStatus.formatted)
[11:50:00|Error] Error: set param busy timeout: CAM_COLORMODE (mavlink_parameters.cpp:477)
failure: setSetting() CameraError(code: Dronecode_SDK_Swift.Camera.CameraResult.Result.error, description: "Error")
[11:50:00|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:222)
[11:50:01|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:222)
[11:50:01|Warn ] sending again, retries to do: 1  (527). (mavlink_commands.cpp:222)
[11:50:02|Error] Retrying failed (527) (mavlink_commands.cpp:239)
[11:50:02|Warn ] Command ack 527 not matching our current command: 525 (mavlink_commands.cpp:143)
[11:50:02|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:222)
[11:50:03|Warn ] sending again, retries to do: 2  (525). (mavlink_commands.cpp:222)
[11:50:03|Warn ] sending again, retries to do: 1  (525). (mavlink_commands.cpp:222)
[11:50:04|Error] Retrying failed (525) (mavlink_commands.cpp:239)
[11:50:04|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:222)
[11:50:05|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:222)
[11:50:05|Warn ] sending again, retries to do: 1  (527). (mavlink_commands.cpp:222)
[11:50:06|Error] Retrying failed (527) (mavlink_commands.cpp:239)
[11:50:06|Info ] heartbeats timed out (system_impl.cpp:311)
[11:50:06|Debug] Lost 3762002188609060911 (mavsdk_impl.cpp:360)
Drone disconnected
[11:50:06|Debug] Discovered 3 component(s) (UUID: 3762002188609060911) (system_impl.cpp:563)
Drone connected with UUID : 3762002188609060911
[11:50:07|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:222)
[11:50:07|Warn ] sending again, retries to do: 2  (525). (mavlink_commands.cpp:222)
[11:50:08|Warn ] sending again, retries to do: 1  (525). (mavlink_commands.cpp:222)
[11:50:08|Error] Retrying failed (525) (mavlink_commands.cpp:239)
[11:50:09|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:222)
[11:50:09|Warn ] sending again, retries to do: 2  (525). (mavlink_commands.cpp:222)
[11:50:10|Warn ] sending again, retries to do: 1  (525). (mavlink_commands.cpp:222)
[11:50:10|Error] Retrying failed (525) (mavlink_commands.cpp:239)
[11:50:11|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:222)
[11:50:12|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:222)
[11:50:12|Warn ] sending again, retries to do: 2  (525). (mavlink_commands.cpp:222)
[11:50:13|Warn ] sending again, retries to do: 1  (525). (mavlink_commands.cpp:222)
[11:50:13|Error] Retrying failed (525) (mavlink_commands.cpp:239)
[11:50:14|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:222)
[11:50:14|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:222)
[11:50:15|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:222)
[11:50:15|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:222)
[11:50:16|Warn ] sending again, retries to do: 1  (527). (mavlink_commands.cpp:222)
[11:50:16|Error] Retrying failed (527) (mavlink_commands.cpp:239)
[11:50:17|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:222)
2019-07-16 11:50:17.874218-0700 DronecodeSDK_Swift_Example[879:207596] Camera Status: CameraStatus(videoOn: false, photoIntervalOn: false, usedStorageMib: 0.0, availableStorageMib: 0.0, totalStorageMib: 0.0, recordingTimeS: 0.0, mediaFolderName: "727E90HD", storageStatus: Dronecode_SDK_Swift.Camera.CameraStatus.StorageStatus.notAvailable)
Camera Status: CameraStatus(videoOn: false, photoIntervalOn: false, usedStorageMib: 0.0, availableStorageMib: 0.0, totalStorageMib: 0.0, recordingTimeS: 0.0, mediaFolderName: "727E90HD", storageStatus: Dronecode_SDK_Swift.Camera.CameraStatus.StorageStatus.notAvailable)
[11:50:18|Warn ] sending again, retries to do: 3  (527). (mavlink_commands.cpp:222)
[11:50:18|Warn ] sending again, retries to do: 2  (527). (mavlink_commands.cpp:222)
[11:50:19|Warn ] sending again, retries to do: 1  (527). (mavlink_commands.cpp:222)
[11:50:19|Error] Retrying failed (527) (mavlink_commands.cpp:239)
[11:50:20|Warn ] sending again, retries to do: 3  (525). (mavlink_commands.cpp:222)
[11:50:21|Warn ] sending again, retries to do: 3  (528). (mavlink_commands.cpp:222)
[11:50:21|Warn ] sending again, retries to do: 2  (528). (mavlink_commands.cpp:222)
[11:50:22|Warn ] sending again, retries to do: 1  (528). (mavlink_commands.cpp:222)
```

douglaswsilva, 2019-07-16:
These are the firmware versions we are using. If you need them to be able to reproduce the issue, I think @byuarus can help providing them (or I can do it, I just need his approval 🙃):
[image]

julianoes, 2019-07-19:
I tried to reproduce this with the [Example with some hacks](https://github.com/mavlink/MAVSDK-Swift-Example/pull/15) but it worked fine.

This was using the iOS simulator and connected directly to the camera Wifi.

Does it happen for you over Wifi directly or just through the ST10C?
And if so could you change my branch so that all I have to do is to press a button to reproduce it?
Thanks.

darioxz, 2019-07-19:
@julianoes Thanks for the example. I will try to build and test it with our internal camera version.

One of our engineers reported a bug in the MAVSDK, but I have to verify this first. I'll keep you posted.

julianoes, 2019-07-19:
@darioxz if you have a link to a pre-release camera update to test against, just let me know.

darioxz, 2019-07-19:
I reproduced the issue with the test app you provided. Thanks! 
I haven't been able to verify the bug reported by one of our engineers, though. 

I will escalate the issue to the camera team. They should be able to give me a clear answer why we have this lockout in our internal camera version and not in the public one.

julianoes, 2019-07-22:
> They should be able to give me a clear answer why we have this lockout in our internal camera version and not in the public one.

It could still be a bug in MAVSDK as well (since it should not lockup either) so if you wanted to provide a pre-release to me directly for debugging that would be good.

julianoes, 2021-03-24:
I hope this is no longer happening, closing.

closed 2021-03-24 by julianoes