mavlink/MAVSDK-Swift #140: Cocoapods

bluecamel, 2019-07-20:
It looks like the podspec is outdated.  I've tried to update it, but can't get XCode to see the module.

Do you have any advice?

Thanks!

bluecamel, 2019-07-20:
I was maybe able to get a bit further by adding a `source_files` entry, but now it can't find the `mavsdk_server` module.  How is that supposed to be installed?  I see in Carthage that it's a binary, but not sure how that would work with cocoapods.

Here's the podspec that I'm using:
```#
# Be sure to run `pod lib lint DroneCore-Swift.podspec'
# to ensure this is a valid spec before submitting.
#
# Any lines starting with a # are optional
# To learn more about a Podspec see http://guides.cocoapods.org/syntax/podspec.html

currentVersion = '0.2.0'

Pod::Spec.new do |spec|
  spec.name             = "MAVSDK-Swift"
  spec.version          = currentVersion
  spec.summary          = "MAVSDK"

# This description is used to generate tags and improve search results.
  spec.description     = "MAVSDK client for Swift"
  spec.homepage        = "https://github.com/dronecore/MAVSDK-Swift"
  spec.license         = { :type => 'BSD', :file => 'LICENSE.md' }
  spec.author          = { "ayameMBS" => "marjory.silvestre@gmail.com" }
  spec.platform        = :ios, '11.0'
  spec.requires_arc    = true
  spec.source          = { :git => '' }
  spec.source_files    = 'MAVSDK_Swift', 'Sources/MAVSDK-Swift/**/*.{h,m,swift}'
  spec.vendored_frameworks = 'mavsdk_server.framework','MAVSDK_Swift.framework'
  spec.dependency 'SwiftGRPC'
  spec.dependency 'RxSwift', '= 5.0.0'
end```

JonasVautherin, 2019-07-20:
Oh, that's super cool, thanks for looking into that!

For `mavsdk_server`, actually if you look at [the file linked in the Cartfile](https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json), you have links to all the released archives of mavsdk_server (used to be called backend).

Taking e.g. the latest one, "mavsdk_server_0.18.3.zip", you will unzip it to `mavsdk_server.framework`, and that is the dependency that is linked in [`project.yml`](https://github.com/mavlink/MAVSDK-Swift/blob/master/project.yml#L21). Note that we use xcodegen to generate the xcodeproj from this `project.yml`.

The previous podspec was actually getting that (or a similar) archive, [see here](https://github.com/mavlink/MAVSDK-Swift/blob/master/DroneCore-Swift.podspec#L22-L23).

Does that help?

bluecamel, 2019-07-20:
Oh, I thought source was an alternate source for the whole repo/build for cocoapods.  Got it.  I'll give that a go and report back.  If I get it working, I'll post a MR.

Also, maybe I should create another issue, but I'll also be needing the offboard protocols.  Any pointers on getting those included?

Thanks!

bluecamel, 2019-07-20:
Hrmm, well, I got past that issue by just downloading and unzipping that file manually.  But now I'm getting lots of compile errors for optionals on the responses from RPC calls.  Maybe I'm dealing with a swift version incompatibility?

Here's an example of one of the compile errors:
```Value of optional type 'Mavsdk_Rpc_Camera_VideoStreamInfoResponse?' must be unwrapped to refer to member 'videoStreamInfo' of wrapped base type 'Mavsdk_Rpc_Camera_VideoStreamInfoResponse'```

JonasVautherin, 2019-07-20:
> I'll also be needing the offboard protocols. Any pointers on getting those included?

Can you open a new issue for that? So I don't forget :-).

> Maybe I'm dealing with a swift version incompatibility?

On which version of Swift are you?

bluecamel, 2019-07-21:
> > I'll also be needing the offboard protocols. Any pointers on getting those included?
> 
> Can you open a new issue for that? So I don't forget :-).
Will do!
> 
> > Maybe I'm dealing with a swift version incompatibility?
> 
> On which version of Swift are you?
We recently converted to Swift 5, but I found out that the conversion was not complete.  I've got that going now (confirmed compile and run without `MAVSDK-Swift` installed), and the app compiles, but I have runtime crashes just by importing `MAVSDK-Swift`.  I'm not evening instantiating `Drone`.

Here's the crash:
```mavsdk_server`pb_field_iter_begin:
    0x1144c3530 <+0>:   pushq  %rbp
    0x1144c3531 <+1>:   movq   %rsp, %rbp
    0x1144c3534 <+4>:   movq   %rdi, -0x8(%rbp)
    0x1144c3538 <+8>:   movq   %rsi, -0x10(%rbp)
    0x1144c353c <+12>:  movq   %rdx, -0x18(%rbp)
    0x1144c3540 <+16>:  movq   -0x10(%rbp), %rdx
    0x1144c3544 <+20>:  movq   -0x8(%rbp), %rsi
    0x1144c3548 <+24>:  movq   %rdx, (%rsi)
    0x1144c354b <+27>:  movq   -0x10(%rbp), %rdx
    0x1144c354f <+31>:  movq   -0x8(%rbp), %rsi
    0x1144c3553 <+35>:  movq   %rdx, 0x8(%rsi)
    0x1144c3557 <+39>:  movq   -0x8(%rbp), %rdx
    0x1144c355b <+43>:  movl   $0x0, 0x10(%rdx)
    0x1144c3562 <+50>:  movq   -0x18(%rbp), %rdx
    0x1144c3566 <+54>:  movq   -0x8(%rbp), %rsi
    0x1144c356a <+58>:  movq   %rdx, 0x18(%rsi)
    0x1144c356e <+62>:  movq   -0x18(%rbp), %rdx
    0x1144c3572 <+66>:  movq   -0x8(%rbp), %rsi
    0x1144c3576 <+70>:  movq   0x8(%rsi), %rsi
->  0x1144c357a <+74>:  movl   0x5(%rsi), %eax
    0x1144c357d <+77>:  movl   %eax, %esi
    0x1144c357f <+79>:  addq   %rsi, %rdx
    0x1144c3582 <+82>:  movq   -0x8(%rbp), %rsi
    0x1144c3586 <+86>:  movq   %rdx, 0x20(%rsi)
    0x1144c358a <+90>:  movq   -0x8(%rbp), %rdx
    0x1144c358e <+94>:  movq   0x20(%rdx), %rdx
    0x1144c3592 <+98>:  movq   -0x8(%rbp), %rsi
    0x1144c3596 <+102>: movq   0x8(%rsi), %rsi
    0x1144c359a <+106>: movslq 0x9(%rsi), %rsi
    0x1144c359e <+110>: addq   %rsi, %rdx
    0x1144c35a1 <+113>: movq   -0x8(%rbp), %rsi
    0x1144c35a5 <+117>: movq   %rdx, 0x28(%rsi)
    0x1144c35a9 <+121>: movq   -0x8(%rbp), %rdx
    0x1144c35ad <+125>: movq   0x8(%rdx), %rdx
    0x1144c35b1 <+129>: cmpl   $0x0, (%rdx)
    0x1144c35b4 <+132>: setne  %cl
    0x1144c35b7 <+135>: andb   $0x1, %cl
    0x1144c35ba <+138>: movzbl %cl, %eax
    0x1144c35bd <+141>: popq   %rbp
    0x1144c35be <+142>: retq   
    0x1144c35bf <+143>: nop ```

I did have to change things in the Podfile as well, as there aren't pods for the versions used in carthage.

Here's the modified Podfile:
```platform :ios, '11.0'

target 'MAVSDK_Swift' do
  use_frameworks!

  $rxVersion = 'https://github.com/ReactiveX/RxSwift.git'

  # Pods for Dronecode-SDK-Swift
  pod 'SwiftGRPC', :git => 'https://github.com/JonasVautherin/grpc-swift.git'
  pod 'RxSwift', :git => $rxVersion

  target 'MAVSDK_SwiftTests' do
    inherit! :search_paths
    # Pods for testing
    pod 'RxBlocking', :git => $rxVersion
    pod 'RxTest', :git => $rxVersion
  end

  target 'MAVSDK_SwiftIntegrationTests' do
    inherit! :search_paths
    # Pods for testing
    pod 'RxBlocking', :git => $rxVersion
    pod 'RxTest', :git => $rxVersion
  end
end

post_install do |installer_representation|
  installer_representation.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['ONLY_ACTIVE_ARCH'] = 'NO'
    end
  end
end```

And the modified podspec:
```#
# Be sure to run `pod lib lint DroneCore-Swift.podspec'
# to ensure this is a valid spec before submitting.
#
# Any lines starting with a # are optional
# To learn more about a Podspec see http://guides.cocoapods.org/syntax/podspec.html

currentVersion = '0.6.0'
mavsdkServerVersion = '0.18.3'

Pod::Spec.new do |spec|
  spec.name             = "MAVSDK-Swift"
  spec.version          = currentVersion
  spec.summary          = "MAVSDK"

# This description is used to generate tags and improve search results.
  spec.description     = "MAVSDK client for Swift"
  spec.homepage        = "https://github.com/dronecore/MAVSDK-Swift"
  spec.license         = { :type => 'BSD', :file => 'LICENSE.md' }
  spec.author          = { "ayameMBS" => "marjory.silvestre@gmail.com" }
  spec.platform        = :ios, '11.0'
  spec.requires_arc    = true
  spec.source          = { :http => 'https://s3.eu-central-1.amazonaws.com/dronecode-sdk/mavsdk_server_#{mavsdkServerVersion}.zip' }
  spec.source_files    = 'MAVSDK_Swift', 'Sources/MAVSDK-Swift/**/*.{h,m,swift}'
  spec.vendored_frameworks = 'mavsdk_server.framework','MAVSDK_Swift.framework'
  spec.dependency 'SwiftGRPC'
  spec.dependency 'RxSwift'
end```

It's crashing with a bad

bluecamel, 2019-07-21:
FWIW, this happens even without importing.  Just installing the pods is enough to make it consistently crash like this on app startup.

I've created a `Podfile` in `MAVSDK-Swift-Example` that installs from my same local `MAVSDK-Swift` pod and it doesn't have the issue, so I guess it's something in my app causing it, but I'm at a loss as to what it could be.  I've been reviewing build settings all evening and nothing is drastically different.  Just a lot more dependencies in my app.

JonasVautherin, 2019-07-21:
>   spec.vendored_frameworks = 'mavsdk_server.framework','MAVSDK_Swift.framework'

Are you sure that `MAVSDK_Swift.framework` is supposed to go there? I must say I'm not used to Cocoapods, but if I look at the [grpc-swift podspec](https://github.com/grpc/grpc-swift/blob/master/SwiftGRPC.podspec), for instance, they don't have any vendored framework. I was thinking that vendored was for `mavsdk_server`, because we distribute a binary. But MAVSDK-Swift can be compiled from sources by Cocoapods, right?

bluecamel, 2019-07-21:
Hrmm, the docs on that don't make that very clear (at least to me).  I have tried leaving only `mavsdk_server.framework` in there and I get the same results.  So, it does appear that `MAVSDK_Swift.framework` is not needed there, but removing it doesn't change anything for me.

Does the crash in `pb_field_iter_begin` give a hint maybe?  I was skimming through the server code and it looks like that would happen when protocols are decoded maybe?  My first question, of course, is why that's running without even being imported, but maybe that's how the server operates, running in a background thread always.  If so, could that suggest that perhaps I'm not properly including the protocol files for it to load?  I'm completely speculating here and grasping at straws.

bluecamel, 2019-07-21:
Actually, looking at the generated pods framework inputs configuration, removing `MAVSDK_Swift.framework` removes `MAVSDK-Swift` from the frameworks copied to the build folder.  So, I think it might be needed, but I have the same crash regardless.

JonasVautherin, 2019-07-21:
I don't get why it would crash at runtime without you doing anything. Starting `mavsdk_server` requires you to call `startMavlink()`, and even instantiating a `Drone()` should not run anything special (but in your case you don't even do that).

Just to be sure I understood correctly: the podspec works when including from the Swift example, but it crashes in your app. Right?

bluecamel, 2019-07-22:
Yep, that's right.  It's bizarre.  I've poured over the linked frameworks and such in the project config after the `pod install` in both projects and they're identical.

bluecamel, 2019-07-22:
Thanks for confirming that it's `startMavlink` that would kick that off.  I'm pretty lost as to how this could be happening.

unipheas, 2019-07-22:
@bluecamel I've documented the process for setting up a new project using Carthage. Maybe it'll help with the setup of Cocoapods. It's all described [here](https://gist.github.com/unipheas/a5f94751c665995c56fba3c3af61a7a0#swift-available-for-macos).

julianoes, 2019-07-22:
@bluecamel I guess you'd need a debug build of all the pieces to hunt down the crash that you're seeing?
My best guess is that one of the libs is missing on load though.

JonasVautherin, 2019-07-22:
@bluecamel: Could you open a draft PR with your podspec, so that I can try it on my side? Just to see if I can reproduce your issue.

bluecamel, 2019-07-22:
> @bluecamel: Could you open a draft PR with your podspec and instructions on how to reproduce, so that I can try it on my side?

Here's a MR with the changes: https://github.com/mavlink/MAVSDK-Swift/pull/143

I'm not so sure about reproduction steps.  I think the key difference is that I'm installing this pod into another pod that's a dependency of my app.  I think that's where things are going sideways on me, but I can't figure out why.

bluecamel, 2019-07-22:
> @bluecamel I guess you'd need a debug build of all the pieces to hunt down the crash that you're seeing?
> My best guess is that one of the libs is missing on load though.

Yeah, I guess I should build `mavsdk_server` myself so that I can debug it?  I've tried zombie detection and address analyzer, but can't get anywhere there.  Admittedly, I'm green on those tools, so could be missing something.

bluecamel, 2019-07-22:
I've tried going back to just install things with carthage, but I guess Swift 5 is not supported.  Does that sound right?

bluecamel, 2019-07-22:
Maybe that's not right.  Carthage isn't happy either way.  Sorry, I'll get back with better info.  I'm pulling my hair out over here just trying to install a package 😭

JonasVautherin, 2019-07-22:
Swift 5 should work. I'm on Swift 5. What's in your Cartfile? Something like `github "mavlink/MAVSDK-Swift" == 0.6.0`?

bluecamel, 2019-07-23:
Thanks so much for your help!  I've abandoned cocoapods for the moment, but I think I see how I could make it work with some more time.  Hopefully I can get back to it in the next week or so.  For now, I've got a combo of cocoapods and carthage working 👍

unipheas, 2019-11-14:
@bluecamel Would you like to close this issue?

julianoes, 2019-11-14:
I guess it's carthage for now until we have Swift PM, right?

JonasVautherin, 2019-11-16:
Yes, I think so. Let's close, we can still reopen later if we see that Cocoapods is needed.

closed 2019-11-16 by JonasVautherin