mavlink/MAVSDK-Swift #144: Cancel mission upload and download now working currently

douglaswsilva, 2019-07-23:
I'm testing the latest SDK with latest firmwares and it seems that this got broken. 
I call the `cancelMissionUpload` command, but it doesn't cancel and there's no callback until the mission finishes uploading. Then I get this:

```
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Debug] Send mission item 87 (mission_impl.cpp:1118)
[05:31:58|Info ] Mission accepted (mission_impl.cpp:165)
17:31:58.041 📰 [DroneIntegrationTestViewController.swift:136 uploadMission()]: +DIT+ Mission Uploaded.
[05:31:58|Warn ] No mission upload in progress (mission_impl.cpp:398)
17:31:58.052 📰 [DroneIntegrationTestViewController.swift:149 cancelMissionUpload()]: +DIT+ Mission Canceled.
```

julianoes, 2019-07-23:
Thanks for the report. I'll check that. Are you testing that against H520 with current firmware?

julianoes, 2019-07-23:
I'm actually quite confused about this one because we have an integration test that tests mission cancel in CI :thinking: .

douglaswsilva, 2019-07-23:
Hm, interesting. This is the code I'm using just as a test, it's a bit messy, but let me know if you can spot anything weird or fell free to just copy and paste in the Swift example app. I have not tested with HITL, so I'll try that now and let you know. 

```
    func testCancelMissionUpload() {
        uploadMission()
        DispatchQueue.main.delay(3) {
            self.cancelMissionUpload()
        }
    }
    
    func createMissionItems() -> [Dronecode_SDK_Swift.Mission.MissionItem] {
        var missionItems: [Dronecode_SDK_Swift.Mission.MissionItem] = [
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.8729565314818, longitudeDeg: -122.30195430771894, relativeAltitudeM: 60.96, speedMS: 4.910375, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: -90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.startPhotoInterval, loiterTimeS: 1.0, cameraPhotoIntervalS: 3.500000238418579),
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.8729565314818, longitudeDeg: -122.303095692281, relativeAltitudeM: 60.96, speedMS: 9.0, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: -90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.stopPhotoInterval, loiterTimeS: .nan, cameraPhotoIntervalS: 3.500000238418579),
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.873197401262615, longitudeDeg: -122.303095692281, relativeAltitudeM: 60.96, speedMS: 4.910375, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: 90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.startPhotoInterval, loiterTimeS: 1.0, cameraPhotoIntervalS: 3.500000238418579),
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.873197401262615, longitudeDeg: -122.30195430771894, relativeAltitudeM: 60.96, speedMS: 9.0, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: 90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.stopPhotoInterval, loiterTimeS: .nan, cameraPhotoIntervalS: 3.500000238418579),
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.87343827025588, longitudeDeg: -122.30195430771894, relativeAltitudeM: 60.96, speedMS: 4.910375, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: -90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.startPhotoInterval, loiterTimeS: 1.0, cameraPhotoIntervalS: 3.500000238418579),
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.87343827025588, longitudeDeg: -122.303095692281, relativeAltitudeM: 60.96, speedMS: 9.0, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: -90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.stopPhotoInterval, loiterTimeS: .nan, cameraPhotoIntervalS: 3.500000238418579),
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.873679138461625, longitudeDeg: -122.303095692281, relativeAltitudeM: 60.96, speedMS: 4.910375, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: 90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.startPhotoInterval, loiterTimeS: 1.0, cameraPhotoIntervalS: 3.500000238418579),
            Dronecode_SDK_Swift.Mission.MissionItem(latitudeDeg: 37.873679138461625, longitudeDeg: -122.30195430771894, relativeAltitudeM: 60.96, speedMS: 9.0, isFlyThrough: false, gimbalPitchDeg: -90.0, gimbalYawDeg: 90.0, cameraAction: Dronecode_SDK_Swift.Mission.MissionItem.CameraAction.stopPhotoInterval, loiterTimeS: .nan, cameraPhotoIntervalS: 3.500000238418579)
        ]
        
        missionItems.append(contentsOf: missionItems)
        
        return missionItems
    }
    
    func uploadMission() {
        TDRDronecodeManager.sharedInstance.drone.mission.uploadMission(missionItems: createMissionItems())
            .subscribeOn(BackgroundScheduler.instance)
            .observeOn(MainScheduler.instance)
            .subscribe(onCompleted: {
                print("+DIT+ Mission Uploaded.")
            }, onError: { error in
                print("+DIT+ Mission not uploaded \(error)")
            })
            .disposed(by: disposeBag)
    }
    
    func cancelMissionUpload() {
        TDRDronecodeManager.sharedInstance.drone.mission.cancelMissionUpload()
            .subscribeOn(BackgroundScheduler.instance)
            .observeOn(MainScheduler.instance)
            .timeout(5, scheduler: MainScheduler.instance)
            .subscribe(onCompleted: {
                print("+DIT+ Mission Canceled.")
            }, onError: { error in
                print("+DIT+ Mission not Canceled \(error).")
            })
            .disposed(by: disposeBag)
    }
```

douglaswsilva, 2019-07-23:
Hm, I wonder if it could be something on the firmware side. I'm going to test it in SITL and see what I get. Thanks for letting me know.

douglaswsilva, 2019-07-23:
Yep, changing the scheduler for `cancelMissionUpload.subscribeOn(scheduler)` fixed it. 
Closing. Thanks for the reply!

closed 2019-07-23 by douglaswsilva