mavlink/MAVSDK-Swift #148: XCTest framework is causing single SIGABRT

unipheas, 2019-08-29:
When building a new iOS project and importing the XCTest.framework into the project it is causing a `signal SIGABRT` error. It seems to be because there is an image missing. This was the output

```
dyld: Library not loaded: @rpath/XCTest.framework/XCTest
Referenced from: /usr/lib/swift/libswiftXCTest.dylib
Reason: image not found
```

JonasVautherin, 2019-08-29:
Oh... are we incorrectly linking XCTest into MAVSDK? I though our dependencies were [those](https://github.com/mavlink/MAVSDK-Swift/blob/master/project.yml#L21-L26) :confused:.

Why is it not an issue with the sample app? Is it because of something we do there?

unipheas, 2019-08-29:
I'll test on 0.6.0 to make sure this is still an issue. I'll let you know tomorrow.

unipheas, 2019-08-29:
After upgrading to 0.6.0 it's the same error. I'm importing all frameworks in the `Carthage/Build/iOS` folder. If not all of them are needed then I can update the instructions to only be [those](https://github.com/mavlink/MAVSDK-Swift/blob/master/project.yml#L21-L26)

unipheas, 2019-09-02:
instructions updated

closed 2019-09-02 by unipheas