mavlink/MAVSDK-Swift #149: Backend framework does not contain bitcode

unipheas, 2019-08-29:
When building a new iOS project and import MAVSDK-Swift, it is failing to build because of an Enabled Bitcode. Here is the output

```
'/Users/unipheas/Developer/MAVSDK-Swift-Starter-App/Carthage/Build/iOS/backend.framework/backend' does not contain bitcode. You must rebuild it with bitcode enabled (Xcode setting ENABLE_BITCODE), obtain an updated library from the vendor, or disable bitcode for this target. file '/Users/unipheas/Developer/MAVSDK-Swift-Starter-App/Carthage/Build/iOS/backend.framework/backend' for architecture arm64
```

As a temporary fix you will need to go into `Build Settings`, click `All`, then search for `bitcode`, then switch `Enable Bitcode` to `No`

JonasVautherin, 2019-08-29:
Using MAVSDK-Swift 0.6.0?

unipheas, 2019-08-29:
Ah no, the README says to use 0.5.0. I'll upgrade and test again. I'll also update the readme. Sorry about that.

JonasVautherin, 2019-08-29:
No worries, good catch on the README!

unipheas, 2019-08-29:
After updating to `0.6.0` I'm not getting the follow bitcode error but for RXTest framework
```
'/Users/unipheas/Developer/MAVSDK-Swift-Starter-App/Carthage/Build/iOS/RxTest.framework/RxTest' does not contain bitcode. You must rebuild it with bitcode enabled (Xcode setting ENABLE_BITCODE), obtain an updated library from the vendor, or disable bitcode for this target. file '/Users/unipheas/Developer/MAVSDK-Swift-Starter-App/Carthage/Build/iOS/RxTest.framework/RxTest' for architecture arm64
```

JonasVautherin, 2019-08-29:
Sounds like `mavsdk_server` is built with bitcode enabled, and `RxTest` isn't. Which is out of our control :-).

closed 2019-08-30 by unipheas