mavlink/MAVSDK-Swift #159: Gimbal `setMode` and `setPitchAndYaw` works but returns `unknown` error

douglaswsilva, 2019-10-23:
**Problem:**
With the builds of `MAVSDK-Swift.framework` **0.7.0** + `mavsdk_server.framework` **0.20.0** Setting gimbal mode or pitchAndYaw both work but return an `unknown` error. 

**Tested with:**
- XCode 10.2.1
- Swift 5
- IOS 12.2
- The drone and camera you already know 

**Repro Steps:**
1. Call `gimbal.setMode` or call`gimbal.setPitchAndYaw` 
2. This is the odd part. You will get an `error` that says **success**, and the mode won't be set.
```
▿ GimbalError
  - code : MAVSDK_Swift.Gimbal.GimbalResult.Result.unknown
  - description : "Success"
```

Let me know if you need more info.

julianoes, 2019-10-24:
That's odd. I looked at the generated files for gimbal but didn't see anything wrong. Can you debug it deeper in the generated files to see what's going on?

And what's the Swift and XCode version? I see some `#if swift(>=4.2)` in `gimbal.pb.swift`.

JonasVautherin, 2019-10-24:
Makes total sense, I don't think it ever worked correctly. Here's why:

[`gimbal.h`](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/gimbal/include/plugins/gimbal/gimbal.h#L42-L45) says:

```cpp
    enum class Result {
        SUCCESS = 0, /**< @brief Success. The gimbal command was accepted. */
        ERROR, /**< @brief Error. An error occured sending the command. */
        TIMEOUT, /**< @brief Timeout. A timeout occured sending the command. */
        UNKNOWN /**< @brief Unspecified error. */
    };
```

[`gimbal.proto`](https://github.com/mavlink/MAVSDK-Proto/blob/master/protos/gimbal/gimbal.proto#L54-L57) says:

```protobuf
    enum Result {
        UNKNOWN = 0; // Unknown error
        SUCCESS = 1; // Command was accepted
        ERROR = 2; // Error occurred sending the command
        TIMEOUT = 3; // Command timed out
    }
```

And [the translation](https://github.com/mavlink/MAVSDK/blob/develop/src/backend/src/plugins/gimbal/gimbal_service_impl.h#L27) is done with:

```cpp
static_cast<rpc::gimbal::GimbalResult::Result>(gimbal_result));
```

I think that the fix is the following: the C++ enum should be modified such that it corresponds to the proto (i.e. `UNKNOWN` should be the first one everywhere).

JonasVautherin, 2019-10-24:
You're right, I changed the proto file when I added the documentation there. Apparently I forgot to propagate the change, I'm sorry :confused:.

douglaswsilva, 2019-11-11:
This PR is meant to solve this: https://github.com/mavlink/MAVSDK/pull/900

closed 2019-11-13 by julianoes