mavlink/MAVSDK-Swift #160: Unable to set camera settings (setSettings)

douglaswsilva, 2019-11-11:
With the latest **Mavlink SDK v0.21.1** and a local built updated version of the **Mavlink-Swift v0.7.0** I am not able to set camera settings. 

**System:**
- Xcode 10.1
- Swift 5
- iPad Mini 4 with IOS 12.2
- I cannot give more details about the drone and camera. =/

I don't get any errors in the logs, aside from a disconnection to the drone. Which makes me think that after trying to set settings, something is blocking the connection channel. This happened via wifi and mfi. 

Upon connection to the app, I get this in the logs:
```
|Info ] 12:02:29|Debug] [12:02:29|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:47)
[12:02:30|Info ] Server started (grpc_server.cpp:12:02:30|Info ] Waiting to discover system...[12:02:30|Info ] New system on: 127.0.0.1:49464 (udp_connection.cpp:263)
[12:02:30|Debug] Component Unsupported component (240) added. (system_impl.cpp:401)
[12:02:30|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:561)
[12:02:30|Info ] System discovered [UUID: 1] (connection_initiator.h:62)
[12:02:30|Info ] New system on: 192.168.42.1:48023 (udp_connection.cpp:263)
[12:02:30|Debug] Component Camera 1 (100) added. (system_impl.cpp:401)
[12:02:31|Debug] request camera info (camera_impl.cpp:140)
[12:02:31|Debug] Component Autopilot (1) added. (system_impl.cpp:401)
[12:02:31|Debug] Component Gimbal (154) added. (system_impl.cpp:401)
[12:02:31|Warn ] sending again, retries to do: 3  (511). (mavlink_commands.cpp:222)
[12:02:31|Error] Error: get param busy timeout: MIS_TAKEOFF_ALT (mavlink_parameters.cpp:456)
[12:02:31|Error] Error: UUID changed (system_impl.cpp:257)
[12:02:31|Warn ] Command ack 520 not matching our current command: 521 (mavlink_commands.cpp:143)
[12:02:31|Info ] Using cached file for Yuneec E90. (camera_impl.cpp:909)
[12:02:31|Debug] step range missing for CAM_CUSTOMWB (camera_definition.cpp:371)
[12:02:31|Error] min range missing for CAM_SPOTAREA (camera_definition.cpp[12:02:31|Warn ] Not found: CAM_SPOTAREA (camera_definition.cpp:220)
[12:02:31|Error] min range missing for CAM_ASPECTRATIO (camera_definition.cpp:345)
[12:02:31|Warn ] Not found: CAM_ASPECTRATIO (camera_definition.cpp:220)
[12:02:31|Debug] Ignoring string params. (camera_definition.cpp:121)
[12:02:31|Warn ] Default missing for CAM_SYSTEMTIME (camera_definition.cpp:196)
[12:02:31|Debug] Ignoring custom params. (camera_definition.cpp:126)
[12:02:31|Debug] Ignoring custom params. (camera_definition.cpp:126)
[12:02:31|Warn ] Default missing for CAM_ZOOM (camera_definition.cpp:196)
[12:02:31|Warn ] Comparison type mismatch between uint8_t and uint16_t (mavlink_parameters.h:407)
[12:02:31|Error] Chosen value smaller than minimum (camera_definition.cpp:550)
```

I imagine it could be related to this issue: https://github.com/mavlink/MAVSDK/commit/08d14225b4bc066a0efb180aed1b44aff04f7379 but I'm not sure. 

Please let me know if you anything anything else from me.

julianoes, 2019-11-12:
I'm looking into it.

julianoes, 2019-11-12:
It looks like I can reproduce this with the C++ integration test:
```
cmake -Bbuild -DSUPERBUILD=Off -DBUILD_BACKEND=Off -DCMAKE_BUILD_TYPE=Debug -H. && cmake --build build -j8 && build/src/integration_tests/integration_tests_runner --gtest_filter="CameraTest.SetSettings"
...
Note: Google Test filter = CameraTest.SetSettings
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from CameraTest
[ RUN      ] CameraTest.SetSettings
[02:34:23|Info ] MAVSDK version: 0.21.1-5-g7cacb3ac-dirty (mavsdk_impl.cpp:25)
[02:34:24|Info ] New system on: 192.168.42.1:54697 (udp_connection.cpp:263)
[02:34:24|Debug] New: System ID: 1 Comp ID: 100 (mavsdk_impl.cpp:390)
[02:34:24|Debug] Component Camera 1 (100) added. (system_impl.cpp:401)
[02:34:24|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:561)
[02:34:24|Debug] Component Autopilot (1) added. (system_impl.cpp:401)
[02:34:24|Debug] MAVLink: info: data link #0 lost (system_impl.cpp:308)
[02:34:24|Debug] MAVLink: info: data link #0 regained (system_impl.cpp:308)
[02:34:24|Debug] Component Gimbal (154) added. (system_impl.cpp:401)
[02:34:25|Debug] request camera info (camera_impl.cpp:140)
[02:34:25|Info ] Using cached file for Yuneec E90. (camera_impl.cpp:909)
[02:34:25|Debug] step range missing for CAM_CUSTOMWB (camera_definition.cpp:371)
[02:34:25|Error] min range missing for CAM_SPOTAREA (camera_definition.cpp:345)
[02:34:25|Warn ] Not found: CAM_SPOTAREA (camera_definition.cpp:220)
[02:34:25|Error] min range missing for CAM_ASPECTRATIO (camera_definition.cpp:345)
[02:34:25|Warn ] Not found: CAM_ASPECTRATIO (camera_definition.cpp:220)
[02:34:25|Debug] Ignoring string params. (camera_definition.cpp:121)
[02:34:25|Warn ] Default missing for CAM_SYSTEMTIME (camera_definition.cpp:196)
[02:34:25|Debug] Ignoring custom params. (camera_definition.cpp:126)
[02:34:25|Debug] Ignoring custom params. (camera_definition.cpp:126)
[02:34:25|Warn ] Default missing for CAM_ZOOM (camera_definition.cpp:196)
[02:34:25|Warn ] Comparison type mismatch between uint8_t and uint16_t (mavlink_parameters.h:407)
[02:34:25|Error] Chosen value smaller than minimum (camera_definition.cpp:550)
[02:34:28|Warn ] Comparison type mismatch between uint8_t and uint16_t (mavlink_parameters.h:407)
[02:34:28|Error] Chosen value smaller than minimum (camera_definition.cpp:550)
[02:34:29|Warn ] Setting DOES_NOT not found. (camera_definition.cpp:759)
[02:34:29|Error] Unknown parameter to get option: DOES_NOT (camera_definition.cpp:605)
[02:34:29|Error] Could not get option value. (camera_impl.cpp:1280)
/home/julianoes/src/MAVSDK/src/integration_tests/camera_test_helpers.cpp:78: Failure
Expected equality of these values:
  status
    Which is: 4-byte object <01-00 00-00>
  std::future_status::ready
    Which is: 4-byte object <00-00 00-00>
/home/julianoes/src/MAVSDK/src/integration_tests/camera_settings.cpp:140: Failure
Expected equality of these values:
  set_setting(camera, "CAM_PHOTOQUAL", "1")
    Which is: 4-byte object <06-00 00-00>
  Camera::Result::SUCCESS
    Which is: 4-byte object <01-00 00-00>

```

julianoes, 2019-11-12:
I have a potential fix. I can cut a patch release once CI is happy in the PR.

closed 2019-11-13 by julianoes