mavlink/MAVSDK-Swift #165: XCode 11.4 Build Error

gregav, 2020-04-08:
I can not build clean project with Carthage:

*** Downloading RxSwift.framework binary at "ShaiTheBravest"
***  Skipped installing RxSwift.framework binary due to the error:
	"Incompatible Swift version - framework was built with 5.0.1 (swiftlang-1001.0.82.4 clang-1001.0.46.5) and the local version is 5.2 (swiftlang-1103.0.32.1 clang-1103.0.32.29)."

I tried with flag: carthage bootstrap --no-use-binaries --platform ios

*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj
Build Failed
	Task failed with exit code 65:
	/usr/bin/xcrun xcodebuild -project ..........

Can you confirm that the MAVSDK-Swift works with the latest XCode, Swift?
And is it possible to make CocoaPods file for this library? Many projects use Pods and they don't mix with the Carthage.

Thank you for your support,
Grega

unipheas, 2020-04-08:
Hi @gregav. I have not confirmed that it works with the latest Xcode. That said, this is an open-source project so our contributions are voluntary and when we have time to do them. If you'd like to contribute to getting Pods to work with this library then it would be much appreciated by the community.

The error with RxSwift shouldn't matter, I've personally seen that error with previous versions. Did you follow the steps within the [README](https://github.com/mavlink/MAVSDK-Swift/blob/master/README.md) to get everything working?

gregav, 2020-04-08:
Hi Brian,
I would love to make Pods but unfortunately I don’t know how to do it.
And I did follow README several times but no success. It just wont build.

unipheas, 2020-04-08:
@gregav Okay. Can you get us the full output of the logs please. I'd follow the instructions step by step and when it does something unexpected you can just stop there. We'd need to see the entire log though to see what's happening.

gregav, 2020-04-08:
grega@DeepBook MAVTest % gem install --user xcodeproj
WARNING:  You don't have /Users/grega/.gem/ruby/2.6.0/bin in your PATH,
	  gem executables will not run.
Successfully installed xcodeproj-1.15.0
Parsing documentation for xcodeproj-1.15.0
Done installing documentation for xcodeproj after 0 seconds
1 gem installed
grega@DeepBook MAVTest % cat Cartfile 
github "mavlink/MAVSDK-Swift" ~> 0.7.0
grega@DeepBook MAVTest % carthage bootstrap --platform ios
*** No Cartfile.resolved found, updating dependencies
*** Fetching MAVSDK-Swift
*** Fetching RxSwift
*** Fetching grpc-swift
*** Downloading binary-only framework backend at "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json"
*** Checking out RxSwift at "5.0.1"
*** Checking out MAVSDK-Swift at "0.7.0"
*** Checking out grpc-swift at "820730e24b9cf035b2889b30d2fe87411e041720"
*** Downloading binary-only framework backend at "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json"
*** xcodebuild output can be found in /var/folders/12/hpqv_jrd5wsdmgct89y18zkw0000gn/T/carthage-xcodebuild.fAOZ8L.log
*** Downloading RxSwift.framework binary at "ShaiTheBravest"
***  Skipped installing RxSwift.framework binary due to the error:
	"Incompatible Swift version - framework was built with 5.0.1 (swiftlang-1001.0.82.4 clang-1001.0.46.5) and the local version is 5.2 (swiftlang-1103.0.32.1 clang-1103.0.32.29)."

    Falling back to building from the source
*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj
Build Failed
	Task failed with exit code 65:
	/usr/bin/xcrun xcodebuild -project /Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/grega/Library/Caches/org.carthage.CarthageKit/DerivedData/11.4_11E146/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/12/hpqv_jrd5wsdmgct89y18zkw0000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO (launched in /Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift)

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/12/hpqv_jrd5wsdmgct89y18zkw0000gn/T/carthage-xcodebuild.fAOZ8L.log

and the xcodebuild log

grega@DeepBook MAVTest % cat /var/folders/12/hpqv_jrd5wsdmgct89y18zkw0000gn/T/carthage-xcodebuild.fAOZ8L.log
/usr/bin/xcrun xcodebuild -project /Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/grega/Library/Caches/org.carthage.CarthageKit/DerivedData/11.4_11E146/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/12/hpqv_jrd5wsdmgct89y18zkw0000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO (launched in /Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift)Command line invocation:
    /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild -project /Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/grega/Library/Caches/org.carthage.CarthageKit/DerivedData/11.4_11E146/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/12/hpqv_jrd5wsdmgct89y18zkw0000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO

User defaults from command line:
    IDEArchivePathOverride = /var/folders/12/hpqv_jrd5wsdmgct89y18zkw0000gn/T/grpc-swift
    IDEDerivedDataPathOverride = /Users/grega/Library/Caches/org.carthage.CarthageKit/DerivedData/11.4_11E146/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720

Build settings from command line:
    CARTHAGE = YES
    CLANG_ENABLE_CODE_COVERAGE = NO
    CODE_SIGN_IDENTITY = 
    CODE_SIGNING_REQUIRED = NO
    GCC_INSTRUMENT_PROGRAM_FLOW_ARCS = NO
    ONLY_ACTIVE_ARCH = NO
    SDKROOT = iphoneos13.4
    SKIP_INSTALL = YES
    STRIP_INSTALLED_PRODUCT = NO

2020-04-08 20:40:07.411 xcodebuild[88369:1050805]  DTDeviceKit: deviceType from 00008030-000468561ED0802E was NULL
2020-04-08 20:40:07.501 xcodebuild[88369:1050812]  DTDeviceKit: deviceType from 00008030-000468561ED0802E was NULL
2020-04-08 20:40:07.772 xcodebuild[88369:1050800]  DTDeviceKit: deviceType from 00008006-001944EC0E47002E was NULL
/Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift: error: manifest parse error(s):
<unknown>:0: warning: using sysroot for 'iPhoneOS' but targeting 'MacOSX'
<unknown>:0: error: unable to load standard library for target 'x86_64-apple-macosx10.10'
fix-carthage-paths.rb:10:in `read': No such file or directory @ rb_sysopen - .build/dependencies-state.json (Errno::ENOENT)
	from fix-carthage-paths.rb:10:in `<main>'
note: Using new build system
note: Building targets in parallel
note: Planning build
note: Constructing build description
[... cut ...] p/MAVTest/Carthage/Checkouts/grpc-swift/.build/checkouts/swift-protobuf/Sources/SwiftProtobuf/JSONMapEncodingVisitor.swift', '/Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift/.build/checkouts/swift-protobuf/Sources/SwiftProtobuf/ZigZag.swift', '/Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift/.build/checkouts/swift-protobuf/Sources/SwiftProtobuf/BinaryEncodingVisitor.swift', '/Users/grega/Desktop/MAVTest/Carthage/Checkouts/grpc-swift/.build/checkouts/swift-protobuf/Sources/Sw

gregav, 2020-04-08:
I have managed to build the framework by changing the Podfile to:

`platform :ios, '9.0'

target 'MAVSDK_Swift' do
  use_frameworks!

  $gRPCVersion = '>= 0.4.2'
  $rxVersion = '>= 4.0'

  # Pods for MAVSDK_Swift
  pod 'SwiftGRPC', $gRPCVersion
  pod 'RxSwift', $rxVersion

  target 'MAVSDK_SwiftTests' do
    inherit! :search_paths
    # Pods for testing
    pod 'RxBlocking', $rxVersion
    pod 'RxTest', $rxVersion
  end

  target 'MAVSDK_SwiftIntegrationTests' do
    inherit! :search_paths
    # Pods for testing
    pod 'RxBlocking', $rxVersion
    pod 'RxTest', $rxVersion
  end
end

post_install do |installer_representation|
  installer_representation.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['ONLY_ACTIVE_ARCH'] = 'NO'
    end
  end
end
`

and manually download the mavsdk_server and add it to the project.
 (https://s3.eu-central-1.amazonaws.com/dronecode-sdk/mavsdk_server_0.22.0.zip)

closed 2020-04-08 by gregav

unipheas, 2020-04-08:
@gregav Thank you. We'll work on updating this library as soon as we can.

JonasVautherin, 2020-04-09:
One thing I'd like to mention is that SPM should soon support binary dependencies (which we need). If I understand correctly, it will come with Swift 5.3, which is in the process of being released. At this point we would possibly drop Carthage and support solely SPM.

The main reason for our struggling with Carthage is that grpc-swift does not support it, so we always had to find workaround, and it keeps breaking regularly.

floridatechnologies, 2020-04-21:
> I have managed to build the framework by changing the Podfile to:
>
> `platform :ios, '9.0'
>
> target 'MAVSDK_Swift' do
> use_frameworks!
>
> $gRPCVersion = '>= 0.4.2'
> $rxVersion = '>= 4.0'
>
> # Pods for [...]

can you specify what you changed?