mavlink/MAVSDK-Swift #166: error Carhfile setup

skibinalexander, 2020-05-02:
*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj
Build Failed
	Task failed with exit code 65:
	/usr/bin/xcrun xcodebuild -project /Users/skibinalexander/Desktop/MAVLinkIOS/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/skibinalexander/Library/Caches/org.carthage.CarthageKit/DerivedData/11.4.1_11E503a/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/g2/brrcmz914ng_50xhbynkv3s80000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO (launched in /Users/skibinalexander/Desktop/MAVLinkIOS/Carthage/Checkouts/grpc-swift)

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/g2/brrcmz914ng_50xhbynkv3s80000gn/T/carthage-xcodebuild.i0xK5N.log
skibinalexander@MacBook-Pro MAVLinkIOS % vim /var/folders/g2/brrcmz914ng_50xhbynkv3s80000gn/T/carthage-xcodebuild.i0xK5N.log

skibinalexander, 2020-05-02:
Help please resolve with problem

julianoes, 2020-05-04:
Please share the steps you were doing, and the output of the log file which is referenced at the bottom, and please put it all between ``` to have it nicely readable here. Thanks.

skibinalexander, 2020-05-04:
Thanks, I solve this problem

closed 2020-05-04 by skibinalexander

JonasVautherin, 2020-05-04:
Great! Would you mind sharing your solution? Maybe it could help when somebody else gets the same issue...