mavlink/MAVSDK-Swift #168: Installation failure

cwkerk, 2020-07-03:
Build failure in the installation process with Xcode Version 11.5 (11E608c):

```
*** Downloading RxSwift.framework binary at "ShaiTheBravest"
***  Skipped installing RxSwift.framework binary due to the error:
	"Incompatible Swift version - framework was built with 5.0.1 (swiftlang-1001.0.82.4 clang-1001.0.46.5) and the local version is 5.2.4 (swiftlang-1103.0.32.9 clang-1103.0.32.53)."

    Falling back to building from the source
*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj
Build Failed
	Task failed with exit code 65:
```

[image]

JonasVautherin, 2020-07-07:
This says that compiling SwiftGRPC failed, but it does not show the error :confused:. You should have a log that comes with it.

AustinRobinson, 2020-07-26:
I am having a similar issue. Was this ever resolved?

julianoes, 2020-07-27:
@cwkerk @AustinRobinson please add the actual log. It says "outut can be found in ... .log". Without that it's hard to know what the problem is.

AustinRobinson, 2020-07-27:
Here is the output of `carthage bootstrap --platform ios`
```console
austinrobinson@Austins-MacBook-Pro mav % carthage bootstrap --platform ios
*** No Cartfile.resolved found, updating dependencies
*** Fetching MAVSDK-Swift
*** Fetching RxSwift
*** Fetching grpc-swift
*** Downloading binary-only framework backend at "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json"
*** Checking out RxSwift at "5.0.1"
*** Checking out grpc-swift at "820730e24b9cf035b2889b30d2fe87411e041720"
*** Checking out MAVSDK-Swift at "0.8.0"
*** Downloading binary-only framework backend at "https://s3.eu-central-1.amazonaws.com/dronecode-sdk/backend.json"
*** xcodebuild output can be found in /var/folders/80/9d7x9hy14xvgh8hb4_tvgjrr0000gn/T/carthage-xcodebuild.BB7B7S.log
*** Downloading RxSwift.framework binary at "ShaiTheBravest"
***  Skipped installing RxSwift.framework binary due to the error:
	"Incompatible Swift version - framework was built with 5.0.1 (swiftlang-1001.0.82.4 clang-1001.0.46.5) and the local version is 5.2.4 (swiftlang-1103.0.32.9 clang-1103.0.32.53)."

    Falling back to building from the source
*** Building scheme "SwiftGRPC-Package" in SwiftGRPC-Carthage.xcodeproj
Build Failed
	Task failed with exit code 65:
	/usr/bin/xcrun xcodebuild -project /Users/austinrobinson/Documents/Xcode\ Projects/mav/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/austinrobinson/Library/Caches/org.carthage.CarthageKit/DerivedData/11.6_11E708/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/80/9d7x9hy14xvgh8hb4_tvgjrr0000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO (launched in /Users/austinrobinson/Documents/Xcode Projects/mav/Carthage/Checkouts/grpc-swift)

This usually indicates that project itself failed to compile. Please check the xcodebuild log for more details: /var/folders/80/9d7x9hy14xvgh8hb4_tvgjrr0000gn/T/carthage-xcodebuild.BB7B7S.log
```

And here are the contents of carthage-xcodebuild.BB7B7S.log
```
/usr/bin/xcrun xcodebuild -project /Users/austinrobinson/Documents/Xcode\ Projects/mav/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/austinrobinson/Library/Caches/org.carthage.CarthageKit/DerivedData/11.6_11E708/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/80/9d7x9hy14xvgh8hb4_tvgjrr0000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO (launched in /Users/austinrobinson/Documents/Xcode Projects/mav/Carthage/Checkouts/grpc-swift)Command line invocation:
    /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild -project "/Users/austinrobinson/Documents/Xcode Projects/mav/Carthage/Checkouts/grpc-swift/SwiftGRPC-Carthage.xcodeproj" -scheme SwiftGRPC-Package -configuration Release -derivedDataPath /Users/austinrobinson/Library/Caches/org.carthage.CarthageKit/DerivedData/11.6_11E708/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720 -sdk iphoneos ONLY_ACTIVE_ARCH=NO CODE_SIGNING_REQUIRED=NO CODE_SIGN_IDENTITY= CARTHAGE=YES archive -archivePath /var/folders/80/9d7x9hy14xvgh8hb4_tvgjrr0000gn/T/grpc-swift SKIP_INSTALL=YES GCC_INSTRUMENT_PROGRAM_FLOW_ARCS=NO CLANG_ENABLE_CODE_COVERAGE=NO STRIP_INSTALLED_PRODUCT=NO

User defaults from command line:
    IDEArchivePathOverride = /var/folders/80/9d7x9hy14xvgh8hb4_tvgjrr0000gn/T/grpc-swift
    IDEDerivedDataPathOverride = /Users/austinrobinson/Library/Caches/org.carthage.CarthageKit/DerivedData/11.6_11E708/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720

Build settings from command line:
    CARTHAGE = YES
    CLANG_ENABLE_CODE_COVERAGE = NO
    CODE_SIGN_IDENTITY = 
    CODE_SIGNING_REQUIRED = NO
    GCC_INSTRUMENT_PROGRAM_FLOW_ARCS = NO
    ONLY_ACTIVE_ARCH = NO
    SDKROOT = iphoneos13.6
    SKIP_INSTALL = YES
    STRIP_INSTALLED_PRODUCT = NO

/var/folders/80/9d7x9hy14xvgh8hb4_tvgjrr0000gn/T/SchemeScriptAction-Htora0: line 2: cd: /Users/austinrobinson/Documents/Xcode: No such file or directory
error: root manifest not found
ruby: No such file or directory -- fix-carthage-paths.rb (LoadError)
note: Using new build system
note: Building targets in parallel
note: Planning build
note: Constructing build description
2020-07-27 09:39:53.552 xcodebuild[34285:517212]  DTDeviceKit: deviceType from 00008006-000F70AC1AC3802E was NULL
2020-07-27 09:39:54.112 xcodebuild[34285:517213]  DTDeviceKit: deviceType from a158f5ba67095e8d0cfc226684b973bb37c6e36b was NULL
CreateBuildDirectory /Users/austinrobinson/Library/Caches/org.carthage.CarthageKit/DerivedData/11.6_11E708/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720/Build/Intermediates.noindex/ArchiveIntermediates/SwiftGRPC-Package/InstallationBuildProductsLocation (in target 'SwiftGRPC' from project 'SwiftGRPC-Carthage')
    cd /Users/austinrobinson/Documents/Xcode\ Projects/mav/Carthage/Checkouts/grpc-swift
    builtin-create-build-directory /Users/austinrobinson/Library/Caches/org.carthage.CarthageKit/DerivedData/11.6_11E708/grpc-swift/820730e24b9cf035b2889b30d2fe87411e041720/Build/Intermediates.noindex/ArchiveIntermediates/Sw [... cut ...] /Sources/SwiftProtobuf/Google_Protobuf_Struct+Extensions.swift', '/Users/austinrobinson/Documents/Xcode Projects/mav/Carthage/Checkouts/grpc-swift/.build/checkouts/swift-protobuf/Sources/SwiftProtobuf/wrappers.pb.swift' (in target 'SwiftProtobuf' from project 'SwiftGRPC-Carthage')

** ARCHIVE FAILED **

The following build commands failed:
	CompileSwiftSources normal armv7 com.apple.xcode.tools.swift.compiler
	CompileSwiftSources normal arm64 com.apple.xcode.tools.swift.compiler
(2 failures)
```

julianoes, 2020-07-28:
Thanks @AustinRobinson. @unipheas have you seen this before?

unipheas, 2020-07-28:
Yea, it looks like the libraries weren't setup within the project. It's in the instructions but basically he'll have to import these manually into the project, I believe. 

http://dronecode-sdk-swift.s3.eu-central-1.amazonaws.com/docs/master/index.html

```
On your application targets’ Build Phases settings tab, click the + icon and choose New Run Script Phase. Create a Run Script in which you specify your shell (ex: /bin/sh), add the following contents to the script area below the shell:
```
Starting here.

AustinRobinson, 2020-07-28:
Does that still apply if I didn't use Carthage? I manually built the framework and copied it over to my project.

unipheas, 2020-08-02:
@AustinRobinson Did that help you?

AustinRobinson, 2020-08-04:
I tried what you said with the same error but in the readme is says to run `carthage bootstrap --platform ios` before doing that.

unipheas, 2020-08-27:
Is it possible to share your project?

JonasVautherin, 2020-11-26:
Closing due to inactivity, and also because we are moving to SwiftPM :tada:.

closed 2020-11-26 by JonasVautherin