mavlink/MAVSDK-Swift #170: Unable to connect to 3DR Solo

AustinRobinson, 2020-07-28:
I am unable to connect to my 3DR Solo. This is the code I am running
```swift
mavsdkServer.run(systemAddress: "udp://0.0.0.0:14550")
self.drone = Drone()
```

And here is the console output that just keeps looping
```
[10:23:33|Info ] MAVSDK version: 0.28.0 (mavsdk_impl.cpp:26)
[10:23:33|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:377)
[10:23:33|Info ] Server started (grpc_server.cpp:43)
[10:23:33|Info ] Server set to listen on 0.0.0.0:63964 (grpc_server.cpp:44)
[10:23:33|Info ] Waiting to discover system on udp://0.0.0.0:14550... (connection_initiator.h:22)
[10:23:33|Info ] New system on: 10.1.1.1:51760 (udp_connection.cpp:256)
[10:23:33|Debug] Component Autopilot (1) added. (system_impl.cpp:352)
[10:23:33|Debug] New: System ID: 10 Comp ID: 0 (mavsdk_impl.cpp:377)
[10:23:33|Warn ] sending again after 0.525962 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:23:34|Warn ] sending again after 1.05177 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:23:34|Warn ] sending again after 1.57237 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[10:23:35|Error] Retrying failed (520) (mavlink_commands.cpp:243)
[10:23:35|Warn ] sending again after 0.522089 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:23:36|Warn ] No UUID received, using system ID instead. (system_impl.cpp:468)
[10:23:36|Debug] Discovered 1 component(s) (UUID: 1) (system_impl.cpp:525)
[10:23:36|Info ] System discovered [UUID: 1] (connection_initiator.h:61)
[10:23:36|Warn ] sending again after 1.02336 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:23:36|Warn ] sending again after 1.53318 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[10:23:37|Debug] Falling back to Gimbal Version 1 (gimbal_impl.cpp:53)
[10:23:37|Warn ] sending again, retries to do: 3  (NAV_MIN_FT_HT). (mavlink_parameters.cpp:458)
[10:23:37|Error] Retrying failed (520) (mavlink_commands.cpp:243)
[10:23:37|Info ] New system on: 10.1.1.1:51760 (udp_connection.cpp:256)
[10:23:37|Warn ] sending again after 0.507869 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:23:38|Warn ] sending again, retries to do: 2  (NAV_MIN_FT_HT). (mavlink_parameters.cpp:458)
[10:23:38|Warn ] sending again after 1.0201 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:23:38|Warn ] sending again after 1.53135 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[10:23:39|Warn ] sending again, retries to do: 1  (NAV_MIN_FT_HT). (mavlink_parameters.cpp:458)
[10:23:39|Error] Retrying failed (520) (mavlink_commands.cpp:243)
[10:23:39|Warn ] sending again after 0.50058 s, retries to do: 3  (511). (mavlink_commands.cpp:224)
[10:23:40|Error] Error: Retrying failed get param busy timeout: NAV_MIN_FT_HT (mavlink_parameters.cpp:474)
[10:23:40|Warn ] sending again after 1.0069 s, retries to do: 2  (511). (mavlink_commands.cpp:224)
[10:23:40|Warn ] sending again after 1.51574 s, retries to do: 1  (511). (mavlink_commands.cpp:224)
[10:23:41|Warn ] sending again, retries to do: 3  (NAV_FT_DST). (mavlink_parameters.cpp:458)
[10:23:41|Error] Retrying failed (511) (mavlink_commands.cpp:243)
[10:23:41|Warn ] sending again after 0.512429 s, retries to do: 3  (512). (mavlink_commands.cpp:224)
[10:23:42|Warn ] sending again, retries to do: 2  (NAV_FT_DST). (mavlink_parameters.cpp:458)
[10:23:42|Warn ] sending again after 1.01987 s, retries to do: 2  (512). (mavlink_commands.cpp:224)
[10:23:42|Warn ] sending again after 1.52798 s, retries to do: 1  (512). (mavlink_commands.cpp:224)
[10:23:43|Warn ] sending again, retries to do: 1  (NAV_FT_DST). (mavlink_parameters.cpp:458)
[10:23:43|Error] Retrying failed (512) (mavlink_commands.cpp:243)
[10:23:43|Warn ] sending again after 0.50097 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:23:44|Error] Error: Retrying failed get param busy timeout: NAV_FT_DST (mavlink_parameters.cpp:474)
[10:23:44|Warn ] sending again after 1.00225 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:23:44|Warn ] sending again after 1.51333 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[10:23:45|Warn ] sending again, retries to do: 3  (NAV_FT_FS). (mavlink_parameters.cpp:458)
[10:23:45|Error] Retrying failed (520) (mavlink_commands.cpp:243)
[10:23:45|Warn ] sending again after 0.511758 s, retries to do: 3  (528). (mavlink_commands.cpp:224)
[10:23:46|Warn ] sending again, retries to do: 2  (NAV_FT_FS). (mavlink_parameters.cpp:458)
[10:23:46|Warn ] sending again after 1.02005 s, retries to do: 2  (528). (mavlink_commands.cpp:224)
[10:23:46|Warn ] sending again after 1.52775 s, retries to do: 1  (528). (mavlink_commands.cpp:224)
[10:23:47|Warn ] sending again, retries to do: 1  (NAV_FT_FS). (mavlink_parameters.cpp:458)
[10:23:47|Error] Retrying failed (528) (mavlink_commands.cpp:243)
[10:23:47|Warn ] sending again after 0.50452 s, retries to do: 3  (528). (mavlink_commands.cpp:224)
[10:23:48|Error] Error: Retrying failed get param busy timeout: NAV_FT_FS (mavlink_parameters.cpp:474)
[10:23:48|Warn ] sending again after 1.01733 s, retries to do: 2  (528). (mavlink_commands.cpp:224)
[10:23:48|Warn ] sending again after 1.51871 s, retries to do: 1  (528). (mavlink_commands.cpp:224)
[10:23:49|Warn ] sending again, retries to do: 3  (NAV_FT_RS). (mavlink_parameters.cpp:458)
[10:23:49|Error] Retrying failed (528) (mavlink_commands.cpp:243)
[10:23:49|Warn ] sending again after 0.501124 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:23:50|Warn ] sending again, retries to do: 2  (NAV_FT_RS). (mavlink_parameters.cpp:458)
[10:23:50|Warn ] sending again after 1.00469 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:23:50|Warn ] sending again after 1.50608 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[... 111 lines cut ...]
[10:24:36|Warn ] sending again after 0.509372 s, retries to do: 3  (528). (mavlink_commands.cpp:224)
[10:24:37|Warn ] sending again after 1.01701 s, retries to do: 2  (528). (mavlink_commands.cpp:224)
[10:24:37|Warn ] sending again after 1.52401 s, retries to do: 1  (528). (mavlink_commands.cpp:224)
[10:24:38|Error] Retrying failed (528) (mavlink_commands.cpp:243)
[10:24:38|Warn ] sending again after 0.505194 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:24:39|Warn ] sending again after 1.01365 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:24:39|Warn ] sending again after 1.5222 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[10:24:40|Error] Retrying failed (520) (mavlink_commands.cpp:243)
[10:24:40|Warn ] sending again after 0.502154 s, retries to do: 3  (528). (mavlink_commands.cpp:224)
[10:24:41|Warn ] sending again after 1.00899 s, retries to do: 2  (528). (mavlink_commands.cpp:224)
[10:24:41|Warn ] sending again after 1.51448 s, retries to do: 1  (528). (mavlink_commands.cpp:224)
[10:24:42|Error] Retrying failed (528) (mavlink_commands.cpp:243)
[10:24:42|Warn ] sending again after 0.502128 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:24:43|Warn ] sending again after 1.00519 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:24:43|Warn ] sending again after 1.51178 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[10:24:44|Error] Retrying failed (520) (mavlink_commands.cpp:243)
[10:24:44|Warn ] sending again after 0.508079 s, retries to do: 3  (528). (mavlink_commands.cpp:224)
[10:24:45|Warn ] sending again after 1.01075 s, retries to do: 2  (528). (mavlink_commands.cpp:224)
[10:24:45|Warn ] sending again after 1.51881 s, retries to do: 1  (528). (mavlink_commands.cpp:224)
[10:24:46|Error] Retrying failed (528) (mavlink_commands.cpp:243)
[10:24:46|Warn ] sending again after 0.505041 s, retries to do: 3  (520). (mavlink_commands.cpp:224)
[10:24:47|Warn ] sending again after 1.01249 s, retries to do: 2  (520). (mavlink_commands.cpp:224)
[10:24:47|Warn ] sending again after 1.5164 s, retries to do: 1  (520). (mavlink_commands.cpp:224)
[10:24:48|Error] Retrying failed (520) (mavlink_commands.cpp:243)
```

unipheas, 2020-07-28:
We'll need some more information please. If you could provide any error logs, what you're running this on, etc that would be more helpful to allow us to help you debug this.

AustinRobinson, 2020-07-28:
I'm running this on an iPhone X running iOS 13.5. Where can I find error logs, I only see what the application outputs on the debug console and included that above.

julianoes, 2020-07-29:
@AustinRobinson one thing to note is that MAVSDK is not currently supporting ArduPilot, only PX4. And I'm assuming you are running ArduCopter on Solo, right?

Also see https://github.com/mavlink/MAVSDK/issues/728.

AustinRobinson, 2020-07-29:
I am running OpenSolo which I believe is the same thing as arducopter at the moment.

julianoes, 2020-07-30:
Right, so unfortunately this is expected, until someone takes up ArduPilot support and maintenance for MAVSDK.

closed 2020-07-30 by julianoes