mavlink/MAVSDK-Swift #171: Shell commands

nab0y4enko, 2020-08-11:
Could you provide example for use shell commands on Shell module?

For example arm/disarm uav using shell.
https://mavlink.io/en/messages/common.html#MAV_CMD_COMPONENT_ARM_DISARM

In my test any command success sent without feedback and actions from UAV's side.  I've tested next commands: `arm`, `/mavsdk.rpc.action.ActionService/Arm`, `mavlink arm`, `commander arm`, `MAV_CMD_COMPONENT_ARM_DISARM 1` and have no any ideas more.

```
drone.shell.send(command: "COMMAND").do(
            onError: { error in
                // Newer called
            },
            onCompleted: {
                // Always calls
            }
        )
        .subscribe()
        .disposed(by: disposeBag)
```

JonasVautherin, 2020-08-12:
Are you running against PX4 running on NuttX? Shell doesn't work with SITL...

nab0y4enko, 2020-08-12:
@JonasVautherin Thanks for help. Yeah, I known about simulation.
[image]

But we also have been testing on Tarot X6 + Herelink RC.

Our development stopped at this point and no one can provide us description/example of simple command we could use with MAVSDK-Swift.

JonasVautherin, 2020-08-12:
We have an example in MAVSDK-C++, it would be nice if you could adapt it to Swift! See [here](https://github.com/mavlink/MAVSDK/blob/develop/examples/mavshell/mavshell.cpp).