mavlink/MAVSDK-Swift #173: Use 2 ESP8266 as Serial Bridge over WIFI

vinura, 2020-10-04:
Is it possible to use two ESP8266 as the transmitter from AIR and Receiver on the ground? I am trying to connect Pixhawk with Tengri GCS(http://tengriuav.com/). I currently use 2 RFDs. Is it possible to use Two ESP8266 ?

JonasVautherin, 2020-10-04:
I really don't know, I don't have much experience with the ESP8266. Maybe somebody here has an idea, but otherwise I think it is more of a general MAVLink/networking question to ask e.g. on the PX4 discuss forum or Slack :+1:.

unipheas, 2020-10-04:
I have some experience with the ESP8266, but it was a bridge from an iOS app to the PX4. I had to write router software onto the ESP8266 so it could route the traffic. I feel like you would have to do a lot of testing to get that to work. Theoretically, I don't see why it can't happen, but I'm not sure if it has been done. Good question though.

vinura, 2020-10-05:
I actually did some cording and was able to parse incoming packets to serial. However, I am stuck in a while loop which parses UDP packets back to serial. Since there is a constant flow of data coming in i am not able to send serial data generated from GCS back to the flight controller.

JonasVautherin, 2020-11-27:
Closing due to inactivity, and also because it is not a MAVSDK problem.

closed 2020-11-27 by JonasVautherin