mavlink/MAVSDK-Swift #184: Cannot find type 'GRPCCallHandler' in scope

seungbeom-son, 2021-02-23:
I followed all steps. But when I tried to build my project, this error appeared.

**Cannot find type 'GRPCCallHandler' in scope**

Is there anything I missed?

[image]

julianoes, 2021-02-23:
@byuarus, @douglaswsilva, or @JonasVautherin?

JonasVautherin, 2021-02-23:
Did you try to run the [example app](https://github.com/mavlink/mavsdk-swift-example) first? So that you could see how it is setup :+1:

seungbeom-son, 2021-02-23:
> Did you try to run the [example app](https://github.com/mavlink/mavsdk-swift-example) first? So that you could see how it is setup 👍

Yes, I have already tried the example app. But the result was same.

byuarus, 2021-02-23:
After updating MAVSDK-Swift in example app I see the same issue.
[image]

JonasVautherin, 2021-02-23:
Hmm I don't understand, I just did it (updated [this line](https://github.com/mavlink/MAVSDK-Swift-Example/blob/master/project.yml#L11) to 0.10.0) and it works for me :confused:.

Did you guys run a clean build?

JonasVautherin, 2021-02-23:
Ok I managed to reproduce it, let me have a look

JonasVautherin, 2021-02-23:
@seungbeom-son: could you try with version 0.10.1?

seungbeom-son, 2021-02-24:
> @seungbeom-son: could you try with version 0.10.1?

0.10.1 works well for me. Building the example app successfully completed. Thank you!!

However, another problem has occured 😭 

Whenever I tried to install the example app after build, Xcode shows 'Unable to install "MAVSDK_Swift_Example"' (No code signature found.). But I think this issue is not related to your SDK. I will try to fix this by myself.

Thank you. 😄

JonasVautherin, 2021-02-24:
> (No code signature found.)

Doesn't it mean that you need to set the Team in the project settings?

seungbeom-son, 2021-02-24:
> > (No code signature found.)
> 
> Doesn't it mean that you need to set the Team in the project settings?

Yes. I set the team in the project settings correctly.

I solved this problem in this way [https://pspdfkit.com/guides/ios/current/knowledge-base/library-not-found-swiftpm/](url)

closed 2021-02-26 by JonasVautherin