mavlink/MAVSDK-Swift #188: Can't make MAVSDK work with our PX4 based drone

viona31, 2021-03-04:
We have a drone which is based on PX4 and esp. 

We already have an Android app which successfully connects to the drone and allows to control the flight:
1) Android attaches to drone's WiFi point
2) In its WiFi network drone's ip is 192.168.4.1
2) Android app connects to drone using 192.168.4.1:8001 udp connection and allows to control the drone.

Also we tried QGroundControl app, in which we create udp connection on 192.168.4.1:8001. The app connects to the drone and allows at least to get some telemetry data (there are some problems with take off however - it does not work).

Now we are trying to make an iOS Swift app to control the drone.
We tried MAVSDKSample app and also tried our own project.

Here is our code and what we get in console for MAVSDK Sample app:
[image]

We get a "No System" error.

It seems that we should somehow set the 192.168.4.1 ip address but we don't know how to properly set it. How to properly do it?

JonasVautherin, 2021-03-04:
The actions are failing because there is "No System", i.e. the drone has not been discovered. You should check for that first. From the log output, we see that a drone has not been detected. When it happens, it looks more like this:

```
[06:30:17|Info ] MAVSDK version: v0.37.0-35-gf192baa8 (mavsdk_impl.cpp:21)
[06:30:17|Debug] New: System ID: 0 Comp ID: 0 (mavsdk_impl.cpp:431)
[06:30:17|Info ] Server started (grpc_server.cpp:45)
[06:30:17|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:46)
[06:30:17|Info ] Waiting to discover system on tcp://3.239.43.170:5790... (connection_initiator.h:21)
[06:30:17|Debug] Component Autopilot (1) added. (system_impl.cpp:324)
[06:30:17|Debug] Component Gimbal (154) added. (system_impl.cpp:324)
[06:30:17|Debug] Component Camera 1 (100) added. (system_impl.cpp:324)
[06:30:17|Debug] Discovered 3 component(s) (UUID: 1) (system_impl.cpp:481)
[06:30:17|Info ] System discovered (connection_initiator.h:57)
```

> Also we tried QGroundControl app, in which we create udp connection on 192.168.4.1:8001. The app connects to the drone

I think that's where the confusion is. You configured QGC to send MAVLink messages to `192.168.4.1`, but MAVSDK actually waits for messages coming from the drone. So you have to configure the drone to send messages to the ip of your app, and tell MAVSDK to listen on that port.

Have a try with [px4-gazebo-headless](https://github.com/jonasvautherin/px4-gazebo-headless): set the app to listen to `udp://:14540`, and point px4-gazebo-headless to it, with something like:

```
docker run --rm -it jonasvautherin/px4-gazebo-headless:1.11.0 192.168.0.12
```

Where 192.168.0.12 would be the IP of the device running your app

viona31, 2021-03-04:
Here is what our drone's hardware developer says:

`Port 8001 is for udp messages exchange. The drone parses their contents and if it detects Mavlink, then it reacts.
Mavlink stands for Ground Control Station (GCS) и Vehicle. 

More often communications is initiated by GCS and а Vehicle produces answer. In this term the drone is more like "server" in terms of client-server architecture (the vehicle is message responder).

In the drone 3 communication patterns are implemented:
1. GCS <—> Vehicle (question / answer)
2. GCS —> Vehicle (for example, the position of flight control sticks - this message does not mean the drone's answer).
3. GCS <— Vehicle (for example, drone's telemetry)`

So how do you think, can we implement all the three patterns with MAVSDK-Swift? How can we configure MAVSDK-Swift to initiate communication with the drone (192.168.4.1:8001)? Is it possible in any way?

JonasVautherin, 2021-03-05:
I agree with you that it would be nice for MAVSDK to initiate the connection, but right now it cannot do it from MAVSDK-Swift. MAVSDK-C++ can, though, and I would like to make it work for MAVSDK-Swift/MAVSDK-Java in the future (probably after [this PR](https://github.com/mavlink/MAVSDK/pull/1331) is finished).

But in the meantime, you'll have to make your drone initiate the connection (that's actually the traditional way it is done in PX4, though I agree with you in that I would rather have the GCS initiate the connection).