mavlink/MAVSDK-Swift #189: Can we use MAVSDK-Swift mavlink message parser separately or see its code?

viona31, 2021-03-11:
Can we use MAVSDK-Swift SDK message parser separately using its code? 

We have a problem as follows. 

We don't get battery information, what we get in SDK listener is:
`Battery(voltageV: 0.0, remainingPercent: -0.01)`

Drone's hardware developer tells that the drone sends a mavlink battery message like this:
`BatteryStatus{id=0, batteryFunction=EnumValue{value=1, entry=MAV_BATTERY_FUNCTION_ALL}, type=EnumValue{value=1, entry=MAV_BATTERY_TYPE_LIPO}, temperature=32767, voltages=null, currentBattery=65, currentConsumed=-1, energyConsumed=-1, batteryRemaining=91, timeRemaining=0, chargeState=EnumValue{value=1, entry=MAV_BATTERY_CHARGE_STATE_OK}}`

So the battery remaining percent is supposed to be in the batteryRemaining mavlink message field. We need to somehow understand if SDK takes the field from the corresponding mavlink message. How can we see that?

Also the drone is manually controled (joystick commands) via "RC_CHANNELS_OVERRIDE" mavlink message. How can we send this field using MAVSDK-Swift?

JonasVautherin, 2021-03-11:
> How can we see that?

That's done in C++, so you have to look [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/telemetry/telemetry_impl.cpp#L1031-L1049). If you want to debug it at the mavlink level, you'll have to do it from C++ :wink:.

> Also the drone is manually controled (joystick commands) via "RC_CHANNELS_OVERRIDE" mavlink message. How can we send this message using MAVSDK-Swift?

We support the `MANUAL_CONTROL` messages in the ManualControl plugin. Why is it better to use `RC_CHANNELS_OVERRIDE` in that case? Can you not use the manual control messages?

viona31, 2021-03-12:
> That's done in C++, so you have to look [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/telemetry/telemetry_impl.cpp#L1031-L1049). If you want to debug it at the mavlink level, you'll have to do it from C++ 😉.

Debugging C++ is definitely not what we want to do 😔

What we found out is that the drone is correctly sending battery info message (message 147 BATTERY_STATUS) with the next payload:
0 = -1
1 = -1
2 = -1
3 = -1
4 = -1
5 = -1
6 = -1
7 = -1
8 = -1
9 = 127
10 = -99
11 = 1
12 = -1
13 = -1
14 = -1
15 = -1
16 = -1
17 = -1
18 = -1
19 = -1
20 = -1
21 = -1
22 = -1
23 = -1
24 = -1
25 = -1
26 = -1
27 = -1
28 = -1
29 = -1
30 = 59
31 = 0
32 = 0
33 = 1
34 = 1
35 = 93
36 = 0
37 = 0
38 = 0
39 = 0
40 = 1

Here, the 36-th byte is 93 which is real battery remaining value and it is correctly parsed on Android (using dronefleet /
mavlink parser). But in Swift we get -0.01 which is "-1" byte divided by 100.

So the question is - how that could happen? Could there be such a rude bug in C++ implementation? Maybe there can be some uncompatibility? 

We know that the drone uses Mavlink2 protocol. Do mav_sdk_server and Swift SDK support mavlink2?

JonasVautherin, 2021-03-12:
> We know that the drone uses Mavlink2 protocol. Do mav_sdk_server and Swift SDK support mavlink2?

Yes that's MAVLink 2, tested against PX4. Are you using a PX4 drone?

> Debugging C++ is definitely not what we want to do :pensive:

I think at least if you look at the code, it will tell you what it does. And that would not be much different if you were manipulating mavlink messages in Swift directly:

```c++
    mavlink_sys_status_t sys_status;
    mavlink_msg_sys_status_decode(&message, &sys_status);

    Telemetry::Battery new_battery;
    new_battery.voltage_v = sys_status.voltage_battery * 1e-3f;
    // FIXME: it is strange calling it percent when the range goes from 0 to 1.
    new_battery.remaining_percent = sys_status.battery_remaining * 1e-2f;
```

So it takes the SYS_STATUS message, and extracts the battery level as `sys_status.battery_remaining * 1e-2f;`

Does that make sense?

viona31, 2021-03-12:
> Are you using a PX4 drone?

I got to know that the drone is PX4-based but not canonical PX4. Sorry for delusion if so.

> So it takes the SYS_STATUS message, and extracts the battery level as `sys_status.battery_remaining * 1e-2f;`

I didn't understand that this is ANOTHER mavlink message that uses the same field. Thank you for pointing that. Now I'm going to ask firmware developer to make changes to the firmware.

Just for the sake of interest - how it is supposed to deal with the fact that BATTERY_STATUS message is described in protocol specification but does not use its battery_remaining field as battery remaining parameter in SDK?

JonasVautherin, 2021-03-12:
> I got to know that the drone is PX4-based but not canonical PX4. Sorry for delusion if so.

I guess I cannot ask which drone that is? :blush:

> how it is supposed to deal with the fact that BATTERY_STATUS message is described in protocol specification

That's a very good point: both [`SYS_STATUS`](http://mavlink.io/en/messages/common.html#SYS_STATUS) and [`BATTERY_STATUS`](http://mavlink.io/en/messages/common.html#BATTERY_STATUS) provide that information. Maybe we should also parse `BATTERY_STATUS`, and somehow choose the correct one for those updates.

@julianoes: do you have an opinion on that?

JonasVautherin, 2021-03-12:
So I think that `telemetry.battery()` should expose a list of batteries (instead of only one) in case there are more than one battery. In the usual case with one battery, that would be a list of length one. Also, MAVSDK should use `BATTERY_STATUS` if it is advertised, but fallback to `SYS_STATUS` if it is not. That's for the MAVSDK part.

@viona31: Now on the drone side, I think that it would make sense for your drone to set the values into `SYS_STATUS` as well, for ground stations that use it (and that includes MAVSDK at this time).

viona31, 2021-03-18:
> I guess I cannot ask which drone that is? 😊

I'm sorry for being wrong. The drone is not based on PX4 autopilot though it tries to implement mavlink. This is a russian drone for educational purposes, I can tell more details in private.

> Now on the drone side, I think that it would make sense for your drone to set the values into `SYS_STATUS` as well, for ground stations that use it (and that includes MAVSDK at this time).

Yes, thank you for helping to find the solution. Though after finding it we faced another trouble with SDK (crash in unknown code place) - we'll open another issue for describing it.

JonasVautherin, 2021-03-18:
Ok, I guess this one is solved then! Feel free to reopen if it is not!

closed 2021-03-18 by JonasVautherin

viona31, 2021-03-19:
We have solved the problem for us by using SYS_STATUS message instead of BATTERY_STATUS but I still think that the issue still exist for the SDK itself. I don't know why but when I press "open" nothing happens

JonasVautherin, 2021-03-21:
> I don't know why but when I press "open" nothing happens

I am not sure I understand. What is that "open" you press? Is it a button in a UI? :thinking:

viona31, 2021-03-22:
> 
> 
> > I don't know why but when I press "open" nothing happens
> 
> I am not sure I understand. What is that "open" you press? Is it a button in a UI? 🤔

Sorry, yes, that was the status of your newly created issue but I thought it's a button to reopen the current one. 

[image]

I agree with this issue https://github.com/mavlink/MAVSDK/issues/1365