mavlink/MAVSDK-Swift #190: Adding support for Catalyst (Running iOS framework on macOS)

rligocki, 2021-03-11:
Hey guys, 

I few weeks ago I started development of custom design ground control station software based on MAVSDK. Our main target is iOS devices. Also I started to think about using Catalyst technology to run this app also on macOS. Problems is that when I enabled this feature in xcode, I get error: 
`
Showing Recent Messages
While building for Mac Catalyst, no library for this platform was found in '/Users/roman/Library/Developer/Xcode/DerivedData/AirUAWARE-fvgjbgnijdphyfgoroqjlvvbhtmf/SourcePackages/artifacts/MAVSDK_Swift/mavsdk_server.xcframework'.
`

From this error message it looks like there is need to build MAVSDK for both architectures and pack it inside xcframework. Does anybody have some experiences in this topic? Maybe this could be done just by modifying github action.

JonasVautherin, 2021-03-11:
Using https://github.com/leetal/ios-cmake/pull/97/files, there may be a way. This could be done at the same time as adding the binary for the M1 platform.

JonasVautherin, 2021-03-11:
@rligocki: do you know how to add catalyst to an xcframework? Is that a a specific `.framework` just for catalyst, or is it a normal `.framework` that does embed catalyst inside?

@kylewludwig: would you have an idea? I'm looking at https://github.com/mavlink/MAVSDK/pull/1355/files#diff-35869851019316cea57b9e7d8c9cf1fb3cc521708fae85bbe2ed5cac42e2c68dR70-R73 but I'm not sure exactly what it means (i.e. if we need MAC, MAC_ARM64, MAC_CATALYST and MAC_ARM64_CATALYST, or just MAC_CATALYST and MAC_ARM64_CATALYST in the xcframework)

JonasVautherin, 2021-03-11:
Oh actually, MAC is a macOS build, and MAC_CATALYST is an iOS build. So I guess that becomes yet another iOS `.framework`. And MAC_ARM64_CATALYST is another iOS `.framework`, this time for ARM64.

So this may be fairly easy to add :+1:.

rligocki, 2021-03-12:
Perfect !!! Thanks for fast response.

In case of building frameworks I have only experience with Carthage. I didn't have to much time during today to do more research, but as you described and said, whole thing might be easy to change. So do you think that it is only necessary to add  new action to build MAC_CATALYST and MAC_ARM64_CATALYST .frameworks and pack them into already existed .xcframework package?

JonasVautherin, 2021-03-12:
Yes, I'll try that once my other PR is merged :+1:. Would be nice if you could test it then, I'll tag you there :blush:

rligocki, 2021-03-12:
Yeap, as soon as it will be available, then I can add new version of MAVSDK to my app. Is it possible to build it with catalyst support by my own?

JonasVautherin, 2021-03-12:
I gave it a quick shot [here](https://github.com/mavlink/MAVSDK/pull/1356), where I got [the following error](https://github.com/mavlink/MAVSDK/pull/1356/checks?check_run_id=2095530981):

```
-- Looking for C++ include stdint.h - found
-- Looking for C++ include stddef.h
-- Looking for C++ include stddef.h - found
-- Check size of lconv
CMake Error at /usr/local/Cellar/cmake/3.19.6/share/cmake/Modules/CheckTypeSize.cmake:133 (try_compile):
  Cannot copy output executable

    ''

  to destination specified by COPY_FILE:

    '/Users/runner/work/MAVSDK/MAVSDK/build/iphoneos/third_party/jsoncpp/jsoncpp/src/jsoncpp-build/CMakeFiles/CheckTypeSize/LCONV_SIZE.bin'

  Unable to find the executable at any of:

    /Users/runner/work/MAVSDK/MAVSDK/build/iphoneos/third_party/jsoncpp/jsoncpp/src/jsoncpp-build/CMakeFiles/CMakeTmp/cmTC_89208
    /Users/runner/work/MAVSDK/MAVSDK/build/iphoneos/third_party/jsoncpp/jsoncpp/src/jsoncpp-build/CMakeFiles/CMakeTmp/Debug/cmTC_89208
    /Users/runner/work/MAVSDK/MAVSDK/build/iphoneos/third_party/jsoncpp/jsoncpp/src/jsoncpp-build/CMakeFiles/CMakeTmp/Debug/cmTC_89208.app/cmTC_89208
    /Users/runner/work/MAVSDK/MAVSDK/build/iphoneos/third_party/jsoncpp/jsoncpp/src/jsoncpp-build/CMakeFiles/CMakeTmp/cmTC_89208.app/cmTC_89208
    /Users/runner/work/MAVSDK/MAVSDK/build/iphoneos/third_party/jsoncpp/jsoncpp/src/jsoncpp-build/CMakeFiles/CMakeTmp/Development/cmTC_89208

Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.19.6/share/cmake/Modules/CheckTypeSize.cmake:264 (__check_type_size_impl)
  src/lib_json/CMakeLists.txt:30 (check_type_size)
```

@rligocki are you willing to debug this? You can use my PR as a starting point: https://github.com/mavlink/MAVSDK/pull/1356. I won't spend more time on this as I don't know what's happening and I don't need Catalyst, but I'm happy to help if you manage to fix that issue :+1:.

rligocki, 2021-03-12:
I will try. As you said, I am the one that want this feature so let me try debug it.

JonasVautherin, 2023-10-11:
I haven't heard of anyone doing it, but I think it should work. At least we do include a macos framework in the xcframework we publish: https://github.com/mavlink/MAVSDK/blob/main/.github/workflows/main.yml#L508-L509

Did you try a macOS project with MAVSDK-Swift already?

Phando, 2023-10-11:
@JonasVautherin Thank you for the quick response. I have made a multiplatform project and added the framework as described at: https://github.com/mavlink/MAVSDK-Swift. Without any additional code, the resulting project builds and runs on iPad but not on macOS due to the signing error. The signing error is most likely caused by the 'Could not resolve framework symlink' warning I am seeing.  That is what led me to questioning the  framework's ability to run on macOS. 

In the file above I see a NO in the build options. Does that need to change to YES? Is there a way to add a target framework to the xcodegen command?

```
  macOS:
    name: ${{ matrix.name }}
    runs-on: macos-12
    strategy:
      matrix:
        include:
          - name: macOS
            build-framework: OFF
          - name: macOS-framework
            build-framework: ON
```

rligocki, 2023-10-11:
What about Catalyst technology. Catalyst allows running iOS apps on macOS

JonasVautherin, 2023-10-12:
> In the file you posted I see a NO in the build options. Does that need to change to YES? Is there a way to add a target framework to the xcodegen command?

Do you mean "ON"? `build-framework: ON`

> Without any additional code, the resulting project builds and runs on iPad but not on macOS due to the signing error.

Does that mean that it does build for a macOS target? That would be something already :blush: 

> What about Catalyst technology. Catalyst allows running iOS apps on macOS

Does MAVSDK-Swift work with Catalyst now? I lost track of that. Anyway it would be nice if it worked for macOS apps, I don't see a reason why not (and we already build the macOS framework).