mavlink/MAVSDK-Swift #191: Mavlink crashes after trying to call «Arm» command

Harvester-76, 2021-03-18:
When I call drone.arm() it completes with an error. After command completion in my «onError» callback I have the next error 

[image]

«The operation couldn’t be completed. (MAVSDK_Swift.Action.ActionError error 1.)» 

After that my application crashes in another thread in binary file (90% that it happens somewhere inside SDK) so that I can’t even understand where exactly it happens. I suppose that there might be some problem in Mavlink communication with the drone but I need to understand what exactly happens and what seems to be wrong for SDK to work correctly.

[image]

JonasVautherin, 2021-03-18:
@douglaswsilva @byuarus: Would you happen to have an opinion on that? Does it look like a crash indeed, or is it because the error is not handled properly? Just to be sure :blush:.

viona31, 2021-03-19:
@JonasVautherin Could you please tell what does "The operation couldn’t be completed. (MAVSDK_Swift.Action.ActionError error 1" error mean? Does this error correspond to MAV_RESULT_FAILED or MAV_RESULT_DENIED?

JonasVautherin, 2021-03-21:
I think your log says `MAVSDK_Swift.Action.ActionResult.Result.unknown`. I would think it is sent from [here](https://github.com/mavlink/MAVSDK/blob/develop/src/plugins/action/action_impl.cpp#L217), so somehow `send_command_async()` fails with an "unknown" code...

Is it a PX4 drone?

viona31, 2021-03-22:
> Is it a PX4 drone?

It's not but it uses mavlink for communication.

Firmware developer says:
For 'arm' command the drone returns https://mavlink.io/en/messages/common.html#MAV_RESULT
either MAV_RESULT_ACCEPTED or MAV_RESULT_FAILED or MAV_RESULT_DENIED.

Can either of this values lead to the error?

JonasVautherin, 2021-03-22:
I am sorry, at this point I think you'll have to go debug what's happening in C++. I only use PX4 myself, and in your case I don't even know what autopilot you are using.