mavlink/MAVSDK-Swift #197: Not able to connect to TXMOD-RFD868.

rligocki, 2021-04-08:
Hey guys, 

currently I work on iOS custom design GCS. With default code it works for SITL simulator hosted in docker: 
```ruby
let port = mavsdkServer.run()
let drone = Drone(port: Int32(port))
```
When I want to do same thing for TXMOD-RFD868 (http://store.rfdesign.com.au/rfd900-txmod-v2-rc-transmitter-module/), it does not work. As I read older issues, currently MAVSDK-Swift does not support communication initialisation. It waits for data from drone side. Problem is, that I am not able to receive any data from TXMOD. Also I would like to ask someone to describe mavsdkServer.run() method and Drone(port:) constructor. What it does and what their parameters means?
 
[image]

JonasVautherin, 2021-04-08:
By default (and that is the way you run it), MAVSDK will listen on udp port 14540. I am not sure how the TXMOD works, but you can try to change that port in MAVSDK:

```swift
let port = mavsdkServer.run(systemAddress: "udp://:14550")
let drone = Drone(port: Int32(port))
```

And also try 14555, but intuitively I would try 14550 first.

rligocki, 2021-04-08:
Hi Jonas, thank you for fast response.

I tried your suggested code but with no luck. Maybe I will be able to debug whole problem with more knowledge. Currently I dont understand what "udp://:14550" does. Does it mean that mavsdkServer waits for incoming MAVLink packets from any devices on port 14550? What is the reason for port that this .run method returns? Does it mean that mavsdkServer receives communication and Drone object connects to this server?

rligocki, 2021-04-08:
In QGroundControl I found in console log this information: "Adding target QHostAddress("192.168.4.1") 14555".

JonasVautherin, 2021-04-08:
> Currently I dont understand what "udp://:14550" does. Does it mean that mavsdkServer waits for incoming MAVLink packets from any devices on port 14550?

Exactly.

> What is the reason for port that this .run method returns?

This is internal to MAVSDK. The swift part (MAVSDK-Swift) talks to `mavsdk_server`, which is in C++. The port there is the one used between MAVSDK-Swift and `mavsdk_server`.

What's important is this `udp://:14540`, which says that your drone sends MAVLink packets on UDP 14540.

> In QGroundControl I found in console log this information: "Adding target QHostAddress("192.168.4.1") 14555".

Yes, that's the other way. I believe that your drone sends to QGC:14550, and receives on Drone:14555. Conversely, QGC listens on 14550, and when it receives messages from the drone, it sees that they come from 14555. So QGC then responds with messages to 14555, which is the port of the drone.

It's easy to get confused at first, but in UDP, both sides have an IP and a port. So QGC has a port on which it listens, and the drone has another port on which it listens.

> I tried your suggested code but with no luck.

Note that you cannot have both QGC and MAVSDK listening on 14550. Can you try to quit QGC and try MAVSDK again?

closed 2021-04-08 by JonasVautherin