mavlink/MAVSDK-Swift #198: MissionItem is missing Actions (takeoff, transition, land)

rligocki, 2021-04-27:
Currently I work on mission planning system and I found, that there is no option to add transition or any other action except creating waypoint. Am I missing something? I have idea to add array of actions with extension that will be executed when waypoint is reached.

JonasVautherin, 2021-04-27:
Yes you're right, a `MissionItem` is defined [here](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission/mission.proto#L164-L185).

Maybe a transition could be added as a `VtolAction`, similarly to the `CameraAction` [here](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission/mission.proto#L172)?

@julianoes what do you think?

rligocki, 2021-04-27:
Or it could be done as array of `Action`. Question is, how to modify MissionItem to make it compatible with MAVSDK C++ version and PX4. This is what I am missing. Bigger picture.

JonasVautherin, 2021-04-27:
> Question is, how to modify MissionItem to make it compatible with MAVSDK C++

The API is defined in the proto file [here](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission/mission.proto#L164-L185), then the C++ headers are auto-generated from that. But we need to find a way to add those features that is compatible with the auto-generation and hence the other supported languages (python, swift, java, go).

> and PX4

The MAVLink logic is implemented manually [here](https://github.com/mavlink/MAVSDK/blob/main/src/plugins/mission/mission_impl.cpp) (inside the framework of those auto-generated headers, of course).

> Or it could be done as array of Action

Yes maybe, what are the available VTOL actions? I'm only aware of the transition ones :slightly_smiling_face:.

For takeoff, why is it not fine to have MAVSDK arm before the mission? Then it will automatically takeoff, right? For landing, we currently have [SetReturnToLaunchAfterMission](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission/mission.proto#L84), I wonder if we could not change that for some `SetBehaviorAfterMission` that would take an enum like `Return`, `Land`, `Hold`...

rligocki, 2021-04-27:
I was thinking about whole mission planning idea same way as it is implemented in QGroundControl. Each step could be waypoint or waypoint with additional action (Takeoff, transition, landing). As you said, landing could be done with `SetBehaviorAfterMission` and I agree. But still I cannot find way to make same behaviour as in QGC.

JonasVautherin, 2021-04-27:
Oh right, then instead of `VtolAction`, actually it could just be an `Action`, that would have the transition, takeoff and land. Just like `CameraAction`, but just `Action`?

rligocki, 2021-04-27:
That was exactly what I was thinking about. But array or single variable of Action class for each waypoint? For simplicity it could be just one action. What about action behaviour? Will it refuse transition before landing? What if I will create Takeoff Action with different position of plane. Will it takeoff on current position and then go to defined position after setpoint altitude is reached?

JonasVautherin, 2021-04-27:
> But array or single variable of Action class for each waypoint? For simplicity it could be just one action.

Agreed, one action per `MissionItem`, just like the `CameraAction` :+1: 

> What if I will create Takeoff Action with different position of plane. Will it takeoff on current position and then go to defined position after setpoint altitude is reached?

This is a PX4/MAVLink question, right? I guess MAVLink does not specify that, and I don't know what PX4 does. But maybe that's not MAVSDK's problem, is it?

> What about action behaviour? Will it refuse transition before landing?

I think that actions for another type of vehicle should just be ignored, no need to refuse the mission. So if your third waypoint is a VTOL transition-to-fw, a VTOL will transition at that point, but a fw or an mc will both ignore that action. This said, it is a PX4 question again, because MAVSDK will just send them. Do you know what PX4 does in such a case?

We will probably need to add a way in MAVSDK to know if the drone is a multicopter, fixedwing or vtol. So that the client code could do something like this (pseudo-code):

```
if (VTOL) {
    addTransitionItem()
}
```

rligocki, 2021-04-27:
So as I understand current CameraAction: After point is reached, MAVSDK sends Action command and autopilot reacts on it?

JonasVautherin, 2021-04-27:
No, the whole mission is uploaded beforehand, just like what QGC does. Under the hood, the `MissionItem` objects are translated to MAVLink messages :+1:. In other words, whenever you upload a mission with MAVSDK, you can download and inspect it in QGC (and the other way round).

rligocki, 2021-04-29:
Ok maybe I understand it now. So additional actions in `MissionItem` will be converted into separate MAVLink message and placed before or after MissionItem waypoint?

JonasVautherin, 2021-04-29:
Yes. I mean, not automatically, you have to implement that part :slightly_smiling_face:. But that's the idea :+1:

rligocki, 2021-04-29:
Perfect, now where exactly new implementation needs to be done? In Swift version of MAVSDK or C++ version?

julianoes, 2021-04-29:
@rligocki have a look here: https://mavsdk.mavlink.io/main/en/cpp/contributing/plugins.html

Of course, for this you would only extend the existing plugin, and not create a new one.

So the first step is here: https://github.com/mavlink/MAVSDK-Proto