mavlink/MAVSDK-Swift #199: MAVSDK structs and classes should conform protocol (Codable, Hashable, Identifiable)

rligocki, 2021-05-24:
During development of iOS app to control PX4 vehicles I found, that there are too many structs and classes that does not conform to protocols like Codable, Hashable, Identifiable. All these protocol must be added explicitly in app code. This solution looks awful. What do you think?

JonasVautherin, 2021-05-24:
All the structs and classes are auto-generated, so it's "just" a matter of changing the templates. If you could give examples of some structs/classes as they are now, and how they should be modified, I can help with the templates :+1:.

I am just not a Swift developer, so I need help on that side :blush:

rligocki, 2022-02-22:
I tried to add Codable protocol to all enums, structs and classes. Problem is that even when I added Codable protocol to templates, there are still some data structures, that are not generated with Codable protocol. One of the problematic class is Mavsdk_Rpc_Core_CoreServiceClient. Where is a template for generation of this class?

JonasVautherin, 2022-02-22:
Is it not generated by gRPC?