mavlink/MAVSDK-Swift #209: Documentation not up to date.

L00rdD, 2022-05-20:
I'm trying to use the MAVSD-SWIFT v1.0.0, but it seems that none of the connection lines described in the README works for me. It seems that the connection system has been updated the *Drone* constructor does not have the  address or port has parameters.

Would it be possible to update the README or provide a link to a up to date documentation ?

JonasVautherin, 2022-05-20:
Does the example help? https://github.com/mavlink/mavsdk-swift-example

L00rdD, 2022-05-29:
Perfect for my needs !!! Would still be great to have an updated Readme so 🤔

If you think the issue is closed it's good for me 👌 thx a lot for your reactivity.

JonasVautherin, 2022-05-29:
Would you be willing to open a PR updating the README? That would be super appreciated :hugs: