mavlink/MAVSDK-Swift #214: Have to connect to QGC first and then run sample app, otherwise telemetry won't display.

mmanionRIIS, 2022-08-09:
When I run the MavSDK sample app, I am able to connect to my mavlink drone. But I’m not able to read much of the telemetry. However, if I connect to QGroundControl first, then close out of QGC and then run the sample app, it does pick up the Telemetry.  Not able to read battery, GPS info, position. But it can read Health, HealthAllOk, and Armed.  We are using "udp://:14550" instead of the default "udp://:14540" if that makes a difference.

Any ideas on what may be causing this? 

_Without connecting to QGC first:_

[image]

_With connecting to QGC first:_

[image]

julianoes, 2022-08-10:
What telemetry does the drone send by default? And is it a PX4 or ArduPilot drone?

mmanionRIIS, 2022-08-10:
It is Ardupilot. I'm not sure what you mean by "send by default". 

The 1st screenshot above shows what happens if you connect without connecting to QGC first, but I'm not sure if that is what you're looking for.

mmanionRIIS, 2022-08-11:
I tried implementing "setRate" functions for GPS, Battery, and Position. I ran it with the setRate in the TelemetryViewModel and in the TelemetryDetailViewModel, adding it into the ObserveTelemetry functions in both cases. No luck. 

I wasn't sure if I was setting the right Hertz so I tried at 1,5,10,100 all different levels. 

[image]
"

julianoes, 2022-08-12:
I'm not sure what to suggest as there are two uncertainties at play, ArduPilot and working with the Swift wrapper. I'm not sure what to suggest except trying it with a C++ example first, or trying it against PX4...

mmanionRIIS, 2022-08-15:
I'll look into trying it with C++ and PX4, thank you for your help!

closed 2022-08-15 by mmanionRIIS