mavlink/MAVSDK-Swift #215: Parameters - setParamFloat works but setParamInt does not

mmanionRIIS, 2022-08-28:
I am able to change the parameters with setParamFloat, but not with setParamInt. I'm thinking it might have to do with using Int32, but thats just speculation. The weird thing is that setParamInt will return "onCompleted" so it will appear that it successfully changes the param, but it does not. Anyone else come across this? 

```
// return Completed and changes the value
    func setParameterFloat(_ parameter: String, _ value: String) {
        if let newValue = Float(value){
            mavsdkDrone.drone?.param.setParamFloat(name: parameter, value: newValue)
                .subscribe(onCompleted: {
                    print("Changed \(parameter) to \(value)")
                }, onError: { (error) in
                    print(error)
                })
                .disposed(by: disposeBag)
            }
    }
    // return Completed but does not change the value
    func setParameterInt(_ parameter: String, _ value: String) {
        if let newValue = Int32(value){
            mavsdkDrone.drone?.param.setParamInt(name: parameter, value: newValue)
                .subscribe(onCompleted: {
                    print("Changed \(parameter) to \(value)")
                }, onError: { (error) in
                    print(error)
                })
                .disposed(by: disposeBag)
            }
    }
    ```

JonasVautherin, 2022-08-28:
What param are you setting? What autopilot (I assume PX4?), and what version of it?

mmanionRIIS, 2022-08-28:
I'm using ArduPilot. I'm unable to change the RTL_LOIT_TIME or RTL_ALT.

I also tried the PILOT_ACCEL_Z,  TERRAIN_SPACING, and STAT_RUNTIME as well, all unsuccessfully.

julianoes, 2022-08-29:
What version of mavsdk_server or MAVSDK-Swift is it using? There have been some fixes, and for ArduPilot there might be some more coming up with MAVSDK v2.0.

mmanionRIIS, 2022-08-29:
I've updated to the latest version of the MAVSDK-Swift on the Main Branch, so 1.1, looks like it was updated 2 months ago.

Drone Specs:
[image]

JonasVautherin, 2022-08-29:
MAVSDK-Swift is on mavsdk-server 1.4.2: https://github.com/mavlink/MAVSDK-Swift/blob/main/Package.swift#L50. Let me try to update that to 1.4.4 tomorrow :+1:

julianoes, 2022-08-30:
Right, so there have been a couple of fixes since:
https://github.com/mavlink/MAVSDK/pull/1703
https://github.com/mavlink/MAVSDK/pull/1788

You can see the changes here: https://github.com/mavlink/MAVSDK/releases

JonasVautherin, 2022-09-01:
Could you try with the newly-published [MAVSDK-Swift v1.1.1](https://github.com/mavlink/MAVSDK-Swift/releases/tag/1.1.1)?

mmanionRIIS, 2022-09-01:
I updated to 1.1.1 and unfortunately had the same result.

JonasVautherin, 2022-09-01:
Any chance you could try to debug what's happening in C++? Two ways:

1. Build MAVSDK-C++ and try it there from your laptop (I guess it's a problem related to Ardupilot, not iOS).
2. Build MAVSDK-C++ with mavsdk_server, run mavsdk_server on your laptop, and connect your iOS app to it using [this interface](https://github.com/mavlink/MAVSDK-Swift/blob/main/Sources/Mavsdk/Drone.swift#L66).

I would personally go for 1, build a small example (e.g. [this one](https://github.com/mavlink/MAVSDK/tree/main/examples/takeoff_and_land)) and try to reproduce there :+1:.

Also maybe @ykhedar has an opinion :innocent:.

mmanionRIIS, 2022-09-01:
Thank you, will try this tomorrow and let you know how it goes!

mmanionRIIS, 2022-09-02:
Still working on resolving some errors I'm getting when building the C++. I'd bet this is on my end and will let you know when I get the C++ app up and running. 

I'm pretty sure my error is related to https://github.com/mavlink/MAVSDK/issues/1875 

[image]

JonasVautherin, 2022-09-02:
Seems like you are trying to build an example from `main` with a MAVSDK from `1.4.x`. How did you install MAVSDK? You should build it from `main` :+1:

mmanionRIIS, 2022-09-06:
I installed MAVSDK on MacOS using Homebrew, following this: https://mavsdk.mavlink.io/main/en/cpp/quickstart.html