mavlink/MAVSDK-Swift #217: How do I connect the camera on the simulator

fanfans1, 2022-10-31:
In QGC you can choose to change the video source and port number to get the emulator video stream, but how do you get it in mavsdk-swift?

JonasVautherin, 2022-10-31:
Which emulator?

MAVSDK only exposes the camera messages. So for instance you can receive the `VIDEO_STREAM_INFORMATION` message that will tell you where a video stream is, and using the library of your choice you can access this video.

fanfans1, 2022-10-31:
Yes, I use Gazebo Simulation. Because we don't have drones yet, we're running on a simulator. I used the MAVSDK-Swift-Example code. Except for the video display function, everything else is normal, and then do not know how to solve. (this is QGC in simulator run: https://docs.px4.io/main/en/simulation/gazebo.html)

JonasVautherin, 2022-10-31:
MAVSDK won't play the video for you. You have to use a Swift library for that. I think that by default Gazebo sends the video as an RTP stream. You need to handle that in your app (e.g. using gstreamer or similar), but that's not involving MAVSDK.

Does that make sense?

fanfans1, 2022-10-31:
I can assume that the current code (using the MAVSDK-Swift-Example method for getting video streams) is fine, mainly because the emulator does not support it, and if you run it on a drone, it will not be a problem. Is that right?

JonasVautherin, 2022-10-31:
> and if you run it on a drone, it will not be a problem

It should not, but it also depends a bit on the drone :see_no_evil:. So you should test the camera against the drone indeed :+1:

fanfans1, 2022-10-31:
All right, thanks

closed 2022-10-31 by JonasVautherin