mavlink/MAVSDK-Swift #220: Hello, when I run Mavsdk-Swift-Example, after connecting the UAV, the CPU will reach 30%. What causes this problem? Can it be solved?

fanfans1, 2022-11-07:


fanfans1, 2022-11-07:
Running Mavsdk-Swift-Example at iOS

JonasVautherin, 2022-11-07:
Difficult to know without knowing which device you are using, which version of MAVSDK, etc.

fanfans1, 2022-11-07:
MAVSDK version: v1.4.2, iPhone X。

JonasVautherin, 2022-11-07:
Interesting. I never profiled it on an iPhone, I don't know what to expect there. Does it consistently stay at 30% during the flight?

fanfans1, 2022-11-07:
Yes, our drone temporarily runs on simulators, sometimes the CPU reaches 50

JonasVautherin, 2022-11-07:
Without the video stream?

fanfans1, 2022-11-07:
No, it contains video streams

JonasVautherin, 2022-11-07:
Can you try without the video streams, for comparison?

fanfans1, 2022-11-07:
I tried, without video streaming CUP is about 2% lower, after changing to the iPhone 11, CUP becomes about 18%

julianoes, 2022-11-07:
Sounds like video streaming is the reason then? Make sure to use video encoding in hardware. Not sure how that works on iOS but I'm sure there are guides.