mavlink/MAVSDK-Swift #221: Is it possible to use Manual Control with an UI joystick?

brunocrocha99, 2022-11-25:
I'm trying to create a simple UI to control simulated and real drones. I can do simple things like taking off and landing, but i wanted to add a joystick, so that the usar can control roll, pitch, throttle and yaw. I saw a manual control example in the mavsdk-python repo, but i try to adapt it in swift, the `drone?.manualControl.setManualControlInput` results on the drone executing the low batter failsafe.

Does this manual control class really require a constant transmition of data? I wish there was more information about it :(

`func armAction() {
        print("armAction")
        
        mavsdk.drone?.action.arm()
            .subscribe {
                MessageViewModel.shared.message = "Armed Success"
            } onError: { (error) in
                MessageViewModel.shared.message = "Error Arming: \(error)"	
            }
            .disposed(by: disposeBag)
        
        //set initial control values
        mavsdk.drone?.manualControl.setManualControlInput(x: 0, y: 0, z: 0.5, r: 0)
            .subscribe {
                MessageViewModel.shared.message = "Manual Control set"
            } onError: { (error) in
                MessageViewModel.shared.message = "Error Manual Control: \(error)"
            }
            .disposed(by: disposeBag)
        
        //start manual control
        mavsdk.drone?.manualControl.startPositionControl()
            .subscribe {
                MessageViewModel.shared.message = "Manual Control started"
            } onError: { (error) in
                MessageViewModel.shared.message = "Error Manual Control: \(error)"
            }
            .disposed(by: disposeBag)
        
    }`

JonasVautherin, 2022-11-25:
There is another example (in C++) [here](https://github.com/mavlink/MAVSDK/blob/main/examples/manual_control/manual_control.cpp#L155), where it is sending the message every 20ms.

> Does this manual control class really require a constant transmition of data?

Long story short: yes :blush:. That's also by design: if your app stops sending updates, we want the autopilot to interpret that as "connection to the RC was lost, time to failsafe", instead of e.g. continuing with the last command you sent until the closest tree :grin:

brunocrocha99, 2022-11-25:
Thank you for the unexpected extra fast reply 😄.

In the C++ example, why is it need to map the throttle values from [-1,1] to [0,1]? The function accepts values lower than 0.5 for downwards throttle?

(I'm not sure if this is the right place to ask this questions, I really just want to have a working iOS MAVLINK GCS 🙏 )

JonasVautherin, 2022-11-25:
> why is it needed to map the throttle values from [-1,1] to [0,1]?

Not entirely sure, but my guess would be that a negative throttle in MAVLink means "rotate the props in the opposite direction". So 0 means "no throttle", and usually you want to go between 0 and 1.

brunocrocha99, 2022-11-25:
Ok, i will close this issue, since my original question is answered. If I have any other questions should I just open another issue or is there any other platform (like slack for example) ?

JonasVautherin, 2022-11-26:
An issue is fine, otherwise there is https://discuss.px4.io/, and there is a Discord at https://discord.gg/Dronecode :+1:

closed 2022-11-26 by brunocrocha99