mavlink/MAVSDK-Swift #224: Unable to Send Optical Flow or Distance Sensor Messages with MAVSDK-Swift

celikemir, 2025-01-26:
Hi,

I am unable to send optical flow or distance sensor messages from mavsdk-swift to the drone. However, I successfully managed to send these messages using pymavlink in Python, as demonstrated in the example code below:

`from pymavlink import mavutil
import time

# USB üzerinden bağlantı
connection = mavutil.mavlink_connection('/dev/tty.usbmodem01', baud=115200)

# Heartbeat bekleyin
connection.wait_heartbeat()

# Mesaj gönder
while True:
    current_distance = 1500  # 1.5 metre
    connection.mav.distance_sensor_send(
        time_boot_ms=int(time.monotonic() * 1000) % 4294967296,
        min_distance=200,
        max_distance=4000,
        current_distance=current_distance,
        type=mavutil.mavlink.MAV_DISTANCE_SENSOR_LASER,  # Doğrudan mavutil üzerinden
        id=0,
        orientation=25,  # DOWNWARD_POINTING
        covariance=255
    )
    time.sleep(0.1)`

Version of MavSdk 1.1.0

I am using example MAVSDK swift sample.

	1.	Is there an example or guide for sending custom MAVLink messages like distance sensor or optical flow via mavsdk-swift?
	2.	Are there specific configurations or prerequisites required in mavsdk-swift for such messages?

julianoes, 2025-01-26:
Sending any message requires MavlinkPassthrough which is not exposed in language wrappers yet. It's planned for v4.

closed 2025-01-26 by julianoes

celikemir, 2025-01-26:
okey. do you have any idea for me ? I would like to bypass and send message px4 via mavlink. I dont have any idea v4 release time.

celikemir, 2025-02-02:
@julianoes   I compiled MAVSDK from scratch for ARM64 iOS devices and obtained mavsdk_server.xcframework, which only contains mavsdk_server_api.h. This API seems to provide only connection functions. I believe I should be able to use the MavlinkPassthrough function in Objective-C within Xcode. What do you think? I am trying to work this on iphone so I am not able to use python and cpp though.

julianoes, 2025-02-02:
Mavsdk_server is a gRPC server. The API is defined in https://github.com/mavlink/MAVSDK-Proto.

You can read more about it here: https://mavsdk.mavlink.io/main/en/cpp/contributing/autogen.html

The current Mavsdk server API does not expose passthrough, as written above.